The-OpenROAD-Project/OpenROAD #7482: Improve user-experience for long global route run failures

oharboe, 2025-06-03:
### Description

untar and run https://drive.google.com/file/d/1Nw162pCjs1HjxjzeW35vl49Qgw9_bp8D/view?usp=sharing

After 7 hours global routing fails, but there's no specific actionable advice in the logs:

```
[ERROR GRT-0116] Global routing finished with congestion. Check the congestion regions in the DRC Viewer.
Elapsed time: 7:06:08[h:]min:sec. CPU time: user 44323.79 sys 870.38 (176%). Peak memory: 58027380KB.
```

[image]

As a user my strategy here would be:

use bazel-orfs to explore some PLACE_DENSITIES, ROUTING_LAYER_ADJUSTMENT values and examine results. This would take a few days though.

I also suspect that improvements in macro placement could help, but I'm unsure how to prod macro placement to create a better starting point for global routing.

### Suggested Solution

Log something at the end that is quantifiable and actionable or if it is hopeless let the user know that they should "give up"(something to the effect: there's no way that this is going to work, you have to increase the area).

oharboe, 2025-06-03:
> Smaller macro halos looks like a place to start to fix this.
> 
> @eder-matheus suggestions on what else we can report?

Can we can your knowledge in C++ and add a warning/info?

grt: try reducing halo from xx to yy.

Also, isnt the halo a global value? Shouldnt it be per group? Auto by default?

eder-matheus, 2025-06-03:
@oharboe GRT doesn't know about halos. This information isn't stored in ODB and AFAIK, only MPL uses it.

@maliberty One thing we can add is a suggested value for the adjustment value (issue https://github.com/The-OpenROAD-Project/OpenROAD/issues/6638). We can give some priority to this task.
If GRT finds that no adjustment value would help in this case, we might thrown a message about increasing die area, but I don't think GRT is able to specify a suggested value for it.

maliberty, 2025-06-03:
Can't you do that with the cone selection today?