The-OpenROAD-Project/OpenROAD #7488: Improve user-experience of macro placement

oharboe, 2025-06-04:
### Description

Untar and run https://drive.google.com/file/d/1B_0RsKZgjLowZlA7cjAqamY0c4zkseZv/view?usp=sharing

The macro placement is not great for [global routing](https://github.com/The-OpenROAD-Project/OpenROAD/issues/7482). 

[image]

Enabling RUDY in the GUI gives an error:

```
[ERROR GRT-0042] Pin auto_buffer_out_a_bits_address[0] does not have geometries in a valid routing layer.
[WARNING GUI-0066] Heat map "Estimated Congestion (RUDY)" has not been populated with data.
```

### Suggested Solution

Output information that the user can use to set up better constraints for the macro placement, such as a recommended halo size. Also, the halo needs to be rearticulated to allow for a different halo of each of the distinct groups of macro types in this case.

eder-matheus, 2025-06-04:
@oharboe RUDY only works when you have at least the pin placement, that's why you got the error. Also, since we don't have the placement for stdcells, the heatmap would be very odd, with specific hotspots that don't translate to the placement that will be generated later in the flow.

maliberty, 2025-06-04:
@gadfort I think there is a general question about heatmap readiness.  Placement density before placement is equally illogical.  Do you have thoughts on a best path to a general mechanism?

maliberty, 2025-06-04:
That sounds reasonable.  We would need some way to indicate a change of state (ie you just placed the design so now placement density is possible).