The-OpenROAD-Project/OpenROAD #7524: Segmentation fault during 3_4_place_resized when sdc file contains a create_generated_clock command

alexroach-rl, 2025-06-06:
### Describe the bug

When a generated clock is specified in the SDC file, a segmentation fault occurs during the 3_4_place_resized stage. It appears to happen during the execution of the report_metrics command in the resize.tcl script.

It happens for both -combinational and -divide_by generated clocks:
For example:
create_generated_clock -name clk_gated -source clk -combinational clkgate/Q
create_generated_clock -name clk_div2 -source clk -divide_by 2 \clk_div2\$_DFF_PN0_/Q

When a generated clock is not specified for this example, the layout builds as expected. 

### Expected Behavior

The expected behaviour is for it to take the generated clocks into account when performing clock tree synthesis.

### Environment

```shell
Unknown git commit, this is not a git repository.

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 6.11.0-24-generic
os: Ubuntu 24.04.2 LTS (Noble Numbat)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:75 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 13.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: d51df1dea57f1bc70b25f46fef5ce7c396d26e99
-- System name: Linux
-- Compiler: GNU 13.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
[... 165 lines cut ...]
 4# sta::Search::findFilteredArrivals(bool) in /radstd/tools/openroad/build/ubuntu24.04/OpenROAD/bin/openroad
 5# sta::Search::findPathEnds(sta::ExceptionFrom*, sta::Vector<sta::ExceptionThru*>*, sta::ExceptionTo*, bool, sta::Corner const*, sta::MinMaxAll const*, unsigned long, unsigned long, bool, float, float, bool, sta::Set<char const*, sta::CharPtrLess>*, bool, bool, bool, bool, bool, bool) in /radstd/tools/openroad/build/ubuntu24.04/OpenROAD/bin/openroad
 6# sta::Sta::findPathEnds(sta::ExceptionFrom*, sta::Vector<sta::ExceptionThru*>*, sta::ExceptionTo*, bool, sta::Corner const*, sta::MinMaxAll const*, int, int, bool, float, float, bool, sta::Set<char const*, sta::CharPtrLess>*, bool, bool, bool, bool, bool, bool) in /radstd/tools/openroad/build/ubuntu24.04/OpenROAD/bin/openroad
 7# 0x000055A6BAF5800C in /radstd/tools/openroad/build/ubuntu24.04/OpenROAD/bin/openroad
 8# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# 0x000074A91BEB86BC in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
11# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
12# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /radstd/tools/openroad/build/ubuntu24.04/OpenROAD/bin/openroad
13# 0x000055A6BAD01FCF in /radstd/tools/openroad/build/ubuntu24.04/OpenROAD/bin/openroad
14# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
15# main in /radstd/tools/openroad/build/ubuntu24.04/OpenROAD/bin/openroad
16# 0x000074A916E2A1CA in /lib/x86_64-linux-gnu/libc.so.6
17# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
18# _start in /radstd/tools/openroad/build/ubuntu24.04/OpenROAD/bin/openroad
Segmentation fault (core dumped)
Command exited with non-zero status 139
Elapsed time: 0:02.33[h:]min:sec. CPU time: user 0.96 sys 0.14 (47%). Peak memory: 135644KB.
```

### Additional Context

[openroad_build.log](https://github.com/user-attachments/files/20622397/openroad_build.log)

[resize_create_generated_clock.tar.gz](https://github.com/user-attachments/files/20622402/resize_create_generated_clock.tar.gz)

precisionmoon, 2025-06-06:
This is an STA issue.   We haven't even got to CTS step (4_1) yet. 

But the SDC seems a little strange:

create_generated_clock -name clk_gated -source clk -combinational clkgate/Q

Usually, clock gater outputs don't become generated clocks.   Whatever clock feeds the clock gater input simply passes through gater output.   Perhaps removing this gen clock alone may resolve the fatal.

maliberty, 2025-06-06:
My bad on the step.  Can this be turned into an sta test case to file upstream?

maliberty, 2025-06-10:
Great.  Once that is fixed we can update the sta version in OR and close this.

maliberty, 2025-06-24:
The stack is still inside sta.  I'm just going to update the sta submodule rather than mess with patch files.  If that doesn't resolve it the problem will have to go back to sta for further fixes.

closed 2025-07-07 by maliberty