The-OpenROAD-Project/OpenROAD #7549: DIEAREA in write_def has lots of redundant ( 0 0 ) instead of just two coordinates

stefanottili, 2025-06-11:
### Describe the bug

grep DIE results/nangate45/gcd/base/*.def
results/nangate45/gcd/base/3_place.def:DIEAREA ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 71900 ) ( 71900 71900 ) ( 71900 0 ) ;
results/nangate45/gcd/base/6_final.def:DIEAREA ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 0 ) ( 0 71900 ) ( 71900 71900 ) ( 71900 0 ) ;

### Expected Behavior

DIEAREA ( 0 0 ) ( 71900 71900 ) ;

### Environment

```shell
M1 MacOS Sequoia, homebrew, OpenROAD v2.0-22095-g9cbf888aac Jun 10, 2025
```

### To Reproduce

cd flow; make; grep DIE results/nangate45/gcd/base/*.def

### Relevant log output

```shell

```

maliberty, 2025-06-11:
@LucasYuki please see if you can reproduce this on linux

maliberty, 2025-06-11:
I just saw the same in another design.  Please fix it.

PR #7563 (2025-06-12): odb: dbIStream clear containers
Closes #7549.

The Odb reader appends the points stored in ODB to the _dbBlock._die_area polygon.

Changes:
- In dbIStream, clean containers before adding values.
closed 2025-06-12 by maliberty