The-OpenROAD-Project/OpenROAD #7581: Dynamic node fails with routing congestion

gadfort, 2025-06-15:
### Describe the bug

Between https://github.com/The-OpenROAD-Project/OpenROAD/compare/4b729ea474e9bd0a3513579b41a8c96fe20dc75e...6e29df0746191b48963a35bf45422d76e2c9922c something changed that is causing a routing congestion issue on a previously working design.
It appears that the change that broke this was after 588bc37d48f2a5ffef6f2b0d93f24e1810a13038

### Expected Behavior

No routing congestion issue

### Environment

```shell
OpenROAD v2.0-22186-g6e29df0746 
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
```

### To Reproduce

[sc_issue_dynamic_node_job0_skywater130_sky130hd_route.global0_20250615-011417.tar.gz](https://github.com/user-attachments/files/20745947/sc_issue_dynamic_node_job0_skywater130_sky130hd_route.global0_20250615-011417.tar.gz)

```
tar xvf sc_issue_dynamic_node_job0_skywater130_sky130hd_route.global0_20250615-011417.tar.gz
cd sc_issue_dynamic_node_job0_skywater130_sky130hd_route.global0_20250615-011417
./run.sh
```

### Relevant log output

```shell
[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[WARNING GRT-0230] Congestion iterations cannot increase overflow, reached the maximum number of times the total overflow can be increased.
[INFO GRT-0197] Via related to pin nodes: 52735
[INFO GRT-0198] Via related Steiner nodes: 2330
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 95509
[INFO GRT-0112] Final usage 3D: 393857

[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
li1                  0             0            0.00%             0 /  0 /  0
met1             30211         25809           85.43%             4 /  1 / 799
met2             38127         36714           96.29%             0 / 10 / 3532
met3             32346         27842           86.08%             4 /  1 / 826
met4             14910         12881           86.39%             0 /  5 / 1358
met5              5112          4084           79.89%             2 /  0 / 140
---------------------------------------------------------------------------------------
Total           120706        107330           88.92%            10 / 17 / 6655

[INFO GRT-0018] Total wirelength: 1045488 um
[INFO GRT-0014] Routed nets: 7246
[ERROR GRT-0116] Global routing finished with congestion. Check the congestion regions in the DRC Viewer.
[ERROR FLW-0001] Global routing failed, saving database to reports/dynamic_node_top_wrap.globalroute-error.odb
Error: sc_global_route.tcl, 64 FLW-0001
```

maliberty, 2025-06-15:
@gadfort sometimes grt is the symptom rather than the cause.  Does anything differ substantially in the good/bad results pre-grt?  grt itself hasn't had a lot of big changes recently.

maliberty, 2025-06-15:
Would you include a working run as well for comparison?

gadfort, 2025-06-15:
Here is a working one from 588bc37: https://drive.google.com/file/d/1cSI4y8LkhVK1xZUtJABpgQRvhxv9pwo4/view?usp=sharing

maliberty, 2025-06-15:
@povik #7485 is your PR so please investigate.  I've started a quick look as well.

povik, 2025-06-15:
@gadfort where can I find a gpl log for the failing run?

gadfort, 2025-06-15:
Log from failing build:
[place.global.log](https://github.com/user-attachments/files/20747594/place.global.log)

Log from working build:
[place.global.log](https://github.com/user-attachments/files/20747590/place.global.log)

povik, 2025-06-16:
Thanks Peter

The first RD in the failing run spikes up the target density:
```
[INFO GPL-0063] New Target Density:             0.9584
```
vs
```
[INFO GPL-0063] New Target Density:             0.6136
```
on the passing run.

Final densities (after all TD/RD) are 0.84 vs 0.69.

The cell areas after gpl are similar:
```
Design area 106462 u^2 44% utilization.
Cell type report:                       Count       Area
  Tap cell                               3202    4006.34
  Timing Repair Buffer                    284    4762.07
  Inverter                                151     844.56
  Sequential cell                        2257   65683.00
  Multi-Input combinational cell         3303   35172.48
  Total                                  9197  110468.45
```
and
```
Design area 107538 u^2 44% utilization.
Cell type report:                       Count       Area
  Tap cell                               3202    4006.34
  Timing Repair Buffer                    951    4827.13
  Inverter                                151     854.57
  Sequential cell                        2257   65683.00
  Multi-Input combinational cell         3303   36173.44
  Total                                  9864  111544.48
```

gadfort, 2025-06-16:
[sc_issue_dynamic_node_job0_skywater130_sky130hd_place.global0_20250616-083250.tar.gz](https://github.com/user-attachments/files/20756949/sc_issue_dynamic_node_job0_skywater130_sky130hd_place.global0_20250616-083250.tar.gz)

@povik  FYI

maliberty, 2025-06-16:
The seems like a very large jump in density.  Perhaps @gudeh work to do filler decreasing instead would help.  I am curious why the density jumps so much in one run vs the other when the final area is so similar.

povik, 2025-06-16:
> On the failing log I noticed this warning on the second timing-driven gpl iteration: [WARNING RSZ-2005] cannot find a viable buffering solution on pin _5181_/Y after 1 rounds of buffering (no solution meets design rules).

That's worth looking at but it's probably not related to the excessive inflation or the congestion.

gudeh, 2025-06-16:
> [sc_issue_dynamic_node_job0_skywater130_sky130hd_place.global0_20250616-083250.tar.gz](https://github.com/user-attachments/files/20756949/sc_issue_dynamic_node_job0_skywater130_sky130hd_place.global0_20250616-083250.tar.gz)
> 
> [@povik](https://github.com/povik) FYI

@gadfort I am getting this error with such tarball: ```
[ERROR ODB-0477] dbTable mask/shift mismatch 127/7 vs 4095/12
[ERROR GUI-0070] ODB-0477
```

maliberty, 2025-06-16:
This file must have been generated in the window in which I broke backward compatibility which I've subsequently fixed (#7589).   To load this particular db you'll need the old bug back.  In dbBlock.h undo:
```
-  dbTable<_dbDft>* _dft_tbl;
+  dbTable<_dbDft, 4096>* _dft_tbl;
```

gudeh, 2025-06-17:
I had to checkout to commit 6e29df0746191b48963a35bf45422d76e2c9922c to reproduce the error.

The working run is for routing, it could be useful to have the working run for global placement.

Here is the final placement, and RUDY at the end of placement:

[image]

[image]

I noticed an unusual initial placement, with already 70% overflow (we usually start with ~100% on ORFS designs):

[image]

It seems we might be calling rsz too early for this design. Because we go from 70% back to 95% overflow (on iteartion 100) after two timing-driven iterations. The usual progression of overflow is from a high value to a low value.

We also seem to have a divergence happening during routability mode.

gadfort, 2025-06-17:
[sc_issue_dynamic_node_job0_skywater130_sky130hd_place.global0_20250617-082138.tar.gz](https://github.com/user-attachments/files/20775208/sc_issue_dynamic_node_job0_skywater130_sky130hd_place.global0_20250617-082138.tar.gz)

@gudeh here is the run the passing commit.

gudeh, 2025-06-20:
This is the failing run during initialPlacement with CG method. I highlighted the nets of three clusters manually.

[image]

Then we run nesterov:

[image]

gudeh, 2025-06-20:
@gadfort, in your flow, does the global placement always start with instances in the middle?

povik, 2025-06-23:
> For example, if we start with a 0.70 overflow (such as this design), we do not run the timing-driven iterations with thresholds higher than 0.70.

That makes sense but it might be good to run the same number of TD iterations just at lower thresholds. E.g. if we start with 0.70 we cap any tresholds at 0.60 to give it a bit of margin. This might become complicated when combined with RD since I think RD uses the assumption no TD iterations happen in the span 0.30-0.60 or similar.

gudeh, 2025-06-23:
yes, RD currently takes hold of 0.60-0.30 range.

gudeh, 2025-06-25:
I ran a test-CI using -skip_initial_place and here is the absolute difference of DPL and DRT wirelength versus master. We get a big increase of wirelength on two designs, no much change on the rest of designs.

[image]

I was talking with @eder-matheus about this today.
[Here are the values in google sheets](https://docs.google.com/spreadsheets/d/1mCfmBRiq7uDSLUkZZO0APikxI0KumwriSkOc8wWfqTo/edit?pli=1&gid=436520289#gid=436520289)

gadfort, 2025-06-27:
@gudeh even with skip-initial-place the design is not routable., so that doesn't do anything useful to unstick me. Maybe:
1. disable the new pass until these issues are resolved
2. provide a method to disable the new pass

either would be able to unstick this.

povik, 2025-06-27:
> Since the first choice didn't solve it

Not sure I follow. What first choice do you mean?

povik, 2025-06-27:
Based on the findings I believe the issue originates in global placement with a non-virtual TD iteration running too early.

Disabling timing-driven; making all TD iterations virtual (`-keep_resize_below_overflow 0.0`); or the first two iterations virtual (`-keep_resize_below_overflow 0.30`) should all be viable workarounds.

gadfort, 2025-07-08:
@gudeh I will correct myself, 2 / 3 designs passed, but once I ran this design with the correct openroad version it still fails in global routing. `v2.0-22706-g9c4e436fd2`. Strangely my heatmap looks nothing like yours 
[image]
I'm still checking, but as far as I can tell this is using timing driven and routability driven options both.

gudeh, 2025-07-08:
It looks different because I had to increase the maximum allowed GPL iterations.

This design showed very small but consistent reductions in congestion during routability and ended up performing an unusually high number of routability iterations, eventually reaching the current limit of 5000 iterations.

I discussed this with @gadfort, and we considered increasing the iteration limit. My idea is to extend the maximum allowed GPL iterations (currently 5000) proportionally to the number of routability iterations. What do you think, @maliberty?

gadfort, 2025-07-08:
@maliberty my suggestion was to revert the iteration number when reverting the routability solutions that would still allow for a 5k limit, but would not be impacted by the number of routability iterations used.

maliberty, 2025-07-08:
I think actually reverting the iteration number would be a bit confusing.  Increasing the limit by the number of reverted iterations make sense.  We should still have some overall cap to prevent endless reverting but it could be a lot higher.

gudeh, 2025-07-10:
Finished as expected

[image]

I just noticed this design oscilates a little on the eariler revert, when still closer to 0.60 overflow, for this reason it seems to consume more nesterov iterations while inside each revert:
```
Iteration | Overflow |     HPWL (um) |  HPWL(%) |   Penalty | Group
---------------------------------------------------------------
     4920 |   0.5917 |  2.522606e+08 |   +2.79% |  2.04e-13 |      
     4930 |   0.5920 |  2.511727e+08 |   -0.43% |  3.01e-13 |      
     4940 |   0.5975 |  2.509997e+08 |   -0.07% |  4.43e-13 |      
     4950 |   0.6024 |  2.506817e+08 |   -0.13% |  6.53e-13 |      
     4960 |   0.6109 |  2.507110e+08 |   +0.01% |  9.62e-13 |      
     4970 |   0.6164 |  2.511540e+08 |   +0.18% |  1.42e-12 |      
     4980 |   0.6166 |  2.510809e+08 |   -0.03% |  2.09e-12 |      
     4990 |   0.6164 |  2.508200e+08 |   -0.10% |  3.08e-12 |      
     5000 |   0.6188 |  2.506571e+08 |   -0.06% |  4.53e-12 |      
     5010 |   0.6162 |  2.505708e+08 |   -0.03% |  6.67e-12 |      
     5020 |   0.6143 |  2.501432e+08 |   -0.17% |  9.83e-12 |      
     5030 |   0.6117 |  2.497819e+08 |   -0.14% |  1.45e-11 |      
     5040 |   0.6112 |  2.493604e+08 |   -0.17% |  2.13e-11 |      
     5050 |   0.6100 |  2.492349e+08 |   -0.05% |  3.14e-11 |      
     5060 |   0.6084 |  2.490044e+08 |   -0.09% |  4.63e-11 |      
     5070 |   0.6044 |  2.491290e+08 |   +0.05% |  6.82e-11 |      
     5080 |   0.5962 |  2.486502e+08 |   -0.19% |  1.00e-10 |      
     5090 |   0.5881 |  2.480571e+08 |   -0.24% |  1.48e-10 |      
     5100 |   0.5777 |  2.475864e+08 |   -0.19% |  2.18e-10 |      
     5110 |   0.5665 |  2.474322e+08 |   -0.06% |  3.21e-10 |      
     5120 |   0.5484 |  2.469139e+08 |   -0.21% |  4.73e-10 |      
     5130 |   0.5281 |  2.463357e+08 |   -0.23% |  6.97e-10 |      
     5140 |   0.5050 |  2.458496e+08 |   -0.20% |  1.03e-09 |      
     5150 |   0.4799 |  2.448794e+08 |   -0.39% |  1.51e-09 |      
     5160 |   0.4559 |  2.443533e+08 |   -0.21% |  2.23e-09 |      
     5170 |   0.4348 |  2.443650e+08 |   +0.00% |  3.28e-09 |      
     5180 |   0.4113 |  2.445598e+08 |   +0.08% |  4.84e-09 |      
     5190 |   0.3851 |  2.444618e+08 |   -0.04% |  7.12e-09 |      
     5200 |   0.3591 |  2.447472e+08 |   +0.12% |  1.05e-08 |      
     5210 |   0.3249 |  2.440774e+08 |   -0.27% |  1.55e-08 |      
     5220 |   0.2974 |  2.442333e+08 |   +0.06% |  2.28e-08 |
```

gudeh, 2025-07-31:
Hi @gadfort, I believe the issues mentioned here have been resolved. Could you please confirm when you get a chance?

gadfort, 2025-07-31:
Works now. Thanks