The-OpenROAD-Project/OpenROAD #7603: global route regression `[ERROR GRT-0232] Routing congestion too high. ` sky130hd/ibex

oharboe, 2025-06-17:
### Describe the bug

untar and run [grt.tar.gz](https://github.com/user-attachments/files/20772401/grt.tar.gz)

```
max displacement           59.3 u
original HPWL          575656.7 u
legalized HPWL         588012.3 u
delta HPWL                    2 %

Took 63 seconds: detailed_placement
global_route -end_incremental -congestion_report_file ./flow/designs/sky130hd/ibex/reports/sky130hd/ibex_core/base/congestion_post_repair_timing.rpt
[ERROR GRT-0232] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: global_route.tcl, 112 GRT-0232
```

### Expected Behavior

Used to work.

### Environment

```shell
v2.0-22251-ge9aef89f7f
```

### To Reproduce

See above

### Relevant log output

```shell

```

oharboe, 2025-06-17:
@jeffng-or @maliberty I figure this design is right on the border of what grt can handle and not an interesting test case as such.

Close?

With this modification, global route works.

```
$ git diff designs/sky130hd/ibex/BUILD.bazel
diff --git a/flow/designs/sky130hd/ibex/BUILD.bazel b/flow/designs/sky130hd/ibex/BUILD.bazel
index 08f7adc3..7335a26f 100644
--- a/flow/designs/sky130hd/ibex/BUILD.bazel
+++ b/flow/designs/sky130hd/ibex/BUILD.bazel
@@ -4,14 +4,14 @@ orfs_flow(
     name = "ibex",
     arguments = {
         "ADDER_MAP_FILE": "",
-        "CORE_UTILIZATION": "45",
+        "CORE_UTILIZATION": "43",
         "PLACE_DENSITY_LB_ADDON": "0.25",
         "TNS_END_PERCENT": "100",
         "REMOVE_ABC_BUFFERS": "1",
         "SYNTH_HDL_FRONTEND": "slang",
         "VERILOG_INCLUDE_DIRS": "flow/designs/src/ibex_sv/vendor/lowrisc_ip/prim/rtl",
-        "CTS_CLUSTER_SIZE": "20",
-        "CTS_CLUSTER_DIAMETER": "50",
+        #"CTS_CLUSTER_SIZE": "20",
+        #"CTS_CLUSTER_DIAMETER": "50",
     },
     pdk = "@docker_orfs//:sky130hd",
     sources = {
```

maliberty, 2025-06-17:
Is this in some way different from what's already passing in the ORFS CI?

closed 2025-06-17 by maliberty