The-OpenROAD-Project/OpenROAD #7616: segfault in macro placement

gadfort, 2025-06-19:
### Describe the bug

When running macro placement on fazyrv after switching to use yosys-slang, the macro placer is segfaulting on this design.

### Expected Behavior

No segfault

### Environment

```shell
OpenROAD v2.0-22133-g4b729ea474 
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
```

### To Reproduce

[sc_issue_fazyrv_job0_ihp130_sg13g2_stdcell_floorplan.macro_placement0_20250619-124132.tar.gz](https://github.com/user-attachments/files/20819662/sc_issue_fazyrv_job0_ihp130_sg13g2_stdcell_floorplan.macro_placement0_20250619-124132.tar.gz)

```
tar xvf sc_issue_fazyrv_job0_ihp130_sg13g2_stdcell_floorplan.macro_placement0_20250619-124132.tar.gz
cd sc_issue_fazyrv_job0_ihp130_sg13g2_stdcell_floorplan.macro_placement0_20250619-124132
./run.sh
```

### Relevant log output

```shell
Die Area: (0.00, 0.00) (1144.60, 293.35),  Floorplan Area: (4.80, 7.56) (1139.52, 287.28)
	Number of std cell instances: 1756
	Area of std cell instances: 27096.23
	Number of macros: 2
	Area of macros: 100978.27
	Halo width: 20.00
	Halo height: 40.00
	Area of macros with halos: 238043.86
	Area of std cell instances + Area of macros: 128074.49
	Floorplan area: 317403.88
	Design Utilization: 0.40
	Floorplan Utilization: 0.13
	Manufacturing Grid: 5

/usr/include/c++/11/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::front() [with _Tp = mpl::Tiling; _Alloc = std::allocator<mpl::Tiling>; std::vector<_Tp, _Alloc>::reference = mpl::Tiling&]: Assertion '!this->empty()' failed.
Signal 6 received
Stack trace:
 0# 0x0000555DECBCBFE2 in /sc_tools/bin/openroad
 1# 0x00007F7F606F8520 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x0000555DECBB6DE8 in /sc_tools/bin/openroad
 6# mpl::HierRTLMP::calculateMacroTilings(mpl::Cluster*) in /sc_tools/bin/openroad
 7# mpl::HierRTLMP::calculateChildrenTilings(mpl::Cluster*) in /sc_tools/bin/openroad
 8# mpl::HierRTLMP::calculateChildrenTilings(mpl::Cluster*) in /sc_tools/bin/openroad
 9# mpl::HierRTLMP::runCoarseShaping() in /sc_tools/bin/openroad
10# mpl::HierRTLMP::run() in /sc_tools/bin/openroad
11# mpl::MacroPlacer::place(int, int, int, int, int, float, int, float, int, int, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, char const*) in /sc_tools/bin/openroad
12# mpl::rtl_macro_placer_cmd(int, int, int, int, float, int, float, int, int, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, char const*) in /sc_tools/bin/openroad
13# 0x0000555DED74DE71 in /sc_tools/bin/openroad
14# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
15# 0x00007F7F64CA6028 in /lib/x86_64-linux-gnu/libtcl8.6.so
16# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
17# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
18# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /sc_tools/bin/openroad
19# 0x0000555DECBCB7E3 in /sc_tools/bin/openroad
20# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
21# main in /sc_tools/bin/openroad
22# 0x00007F7F606DFD90 in /lib/x86_64-linux-gnu/libc.so.6
23# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
24# _start in /sc_tools/bin/openroad
```

joaomai, 2025-06-19:
The problem here rises from MPL simulated annealing being unable to find a placement that can fit the two macros within the outline, and from what I could gather it is actually impossible:
- Both macros are 824.48 x 144.36 (halos included)
- The floorplaning area is 1134.72 x 279.72
Are those the expected sizes for the macros and floorplaning area? Reducing the halo_height from 40 to 37 makes MPL finish without errors.

Even though it's impossible to find a solution, it should error by providing some explanation instead of simply segfaulting.

gadfort, 2025-06-19:
@joaomai it's possible with the change in the yosys frontend, the synthesis result is a little smaller, triggering this issue. Thanks for diagnosing. I'll correct the size of the layout.

joaomai, 2025-06-19:
I'll open a PR with an improvement to clarify what's the problem if it ever happens again.

PR #7624 (2025-06-19): mpl: error if there are no valid tilings
Closes #7616.

The check that threw an error was on the wrong place. If no valid tilings are found now, it will throw an error instead of segfaulting.
closed 2025-06-19 by maliberty