The-OpenROAD-Project/OpenROAD #7664: gpl: resizer crash

donn, 2025-06-26:
### Describe the bug

The resizer crashes in `gpl`.

Issue bisects to c742efd6ef942f8b12c6a8ca3b5160c82b7974d8

### Expected Behavior

No crash

### Environment

```shell
kernel: Darwin 24.5.0
os: macOS 15.5
cmake version 3.31.6
CMake Deprecation Warning at third-party/abc/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

CMake Warning (dev) at src/CMakeLists.txt:138 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at src/CMakeLists.txt:229 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=TRUE

CMake Warning (dev) at src/dpl/CMakeLists.txt:6 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/cts/CMakeLists.txt:4 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/drt/CMakeLists.txt:17 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/utl/CMakeLists.txt:6 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/dst/CMakeLists.txt:12 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/dft/test/CMakeLists.txt:19 (find_package):
[... 259 lines cut ...]
 8# _ZZN3rsz8Rebuffer18annotateLoadSlacksERNSt3__110shared_ptrINS_11BufferedNetEEEPN3sta6VertexEENK3$_0clINS_7visitorIS9_EEEEiRT_iRKS4_ in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
 9# _ZZN3rsz8Rebuffer18annotateLoadSlacksERNSt3__110shared_ptrINS_11BufferedNetEEEPN3sta6VertexEENK3$_0clINS_7visitorIS9_EEEEiRT_iRKS4_ in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
10# _ZZN3rsz8Rebuffer18annotateLoadSlacksERNSt3__110shared_ptrINS_11BufferedNetEEEPN3sta6VertexEENK3$_0clINS_7visitorIS9_EEEEiRT_iRKS4_ in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
11# _ZZN3rsz8Rebuffer18annotateLoadSlacksERNSt3__110shared_ptrINS_11BufferedNetEEEPN3sta6VertexEENK3$_0clINS_7visitorIS9_EEEEiRT_iRKS4_ in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
12# _ZZN3rsz8Rebuffer18annotateLoadSlacksERNSt3__110shared_ptrINS_11BufferedNetEEEPN3sta6VertexEENK3$_0clINS_7visitorIS9_EEEEiRT_iRKS4_ in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
13# _ZZN3rsz8Rebuffer18annotateLoadSlacksERNSt3__110shared_ptrINS_11BufferedNetEEEPN3sta6VertexEENK3$_0clINS_7visitorIS9_EEEEiRT_iRKS4_ in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
14# rsz::Rebuffer::fullyRebuffer(sta::Pin*) in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
15# rsz::Resizer::findResizeSlacks(bool) in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
16# gpl::TimingBase::executeTimingDriven(bool) in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
17# gpl::NesterovPlace::doNesterovPlace(int) in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
18# gpl::Replace::doNesterovPlace(int, int) in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
19# _wrap_replace_nesterov_place_cmd(void*, Tcl_Interp*, int, Tcl_Obj* const*) in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
20# Tcl_EvalObjv in /nix/store/3rx7vmr58q9nspp79f85na5cnrpfv932-tcl-8.6.15/lib/libtcl8.6.dylib
21# TclEvalEx in /nix/store/3rx7vmr58q9nspp79f85na5cnrpfv932-tcl-8.6.15/lib/libtcl8.6.dylib
22# Tcl_Eval in /nix/store/3rx7vmr58q9nspp79f85na5cnrpfv932-tcl-8.6.15/lib/libtcl8.6.dylib
23# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
24# ord::tclInit(Tcl_Interp*) in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
25# Tcl_MainEx in /nix/store/3rx7vmr58q9nspp79f85na5cnrpfv932-tcl-8.6.15/lib/libtcl8.6.dylib
26# main in /nix/store/pjhjxzbjwpbj8nagi39ppz0ym4jc75hz-openroad-2025-06-26/bin/.openroad-wrapped
./run.sh: line 14: 71550 Segmentation fault: 11  openroad -exit -no_splash -metrics ./or_metrics_out.json ./scripts/openroad/gpl.tcl
```

### Additional Context

Modified Env.sh as usual to use my cmake flags

povik, 2025-06-27:
> Based on the bisect it seems related to https://github.com/The-OpenROAD-Project/OpenROAD/issues/7581

At least in the run sent by @donn (https://github.com/librelane/librelane/actions/runs/15912209330/job/44884710161) the symptoms are different from #7581 in that the placement density at end of gpl doesn't spike up.

povik, 2025-06-27:
@donn I'm interested in QoR degradations. I'll take a look at the failing designs on your CI link

closed 2025-06-28 by maliberty