The-OpenROAD-Project/OpenROAD #7684: Assert happening during detailed routing

titan73, 2025-06-30:
### Describe the bug

Error: pnr.tcl, 126 vector::_M_range_check: __n (which is 9) >= this->size() (which is 9)

[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
    Completing 10% with 0 violations.
    elapsed time = 00:01:24, memory = 1015.13 (MB).
    Completing 20% with 0 violations.
    elapsed time = 00:03:12, memory = 1128.25 (MB).
Error: pnr.tcl, 126 vector::_M_range_check: __n (which is 9) >= this->size() (which is 9)
openroad> 

The version used is OpenROAD v2.0-20726-g7071d85da6 as I cannot test with latest OR due to bug https://github.com/issues/created?issue=The-OpenROAD-Project%7COpenROAD%7C7667.

### Expected Behavior

No assert should happen.

### Environment

```shell
OpenROAD v2.0-20726-g7071d85da6
```

### To Reproduce

I've uploaded a test case full_test_case_2025-06-30.zip on PII server.

### Relevant log output

```shell
[INFO DRT-0036] poly guide region query size = 0.
[INFO DRT-0036] contact guide region query size = 0.
[INFO DRT-0036] metal1 guide region query size = 61312.
[INFO DRT-0036] v1 guide region query size = 0.
[INFO DRT-0036] metal2 guide region query size = 56642.
[INFO DRT-0036] v2 guide region query size = 0.
[INFO DRT-0036] metal3 guide region query size = 27028.
[INFO DRT-0036] v3tk guide region query size = 0.
[INFO DRT-0036] THICK_CU_D guide region query size = 2.
[INFO DRT-0179] Init gr pin query.
[INFO DRT-0267] cpu time = 00:00:01, elapsed time = 00:00:01, memory = 761.76 (MB), peak = 761.76 (MB)
[INFO DRT-0245] skipped writing guide updates to database.
[INFO DRT-0185] Post process initialize RPin region query.
[INFO DRT-0181] Start track assignment.
[INFO DRT-0184] Done with 56644 vertical wires in 5 frboxes and 88340 horizontal wires in 3 frboxes.
[INFO DRT-0186] Done with 19256 vertical wires in 5 frboxes and 15335 horizontal wires in 3 frboxes.
[INFO DRT-0182] Complete track assignment.
[INFO DRT-0267] cpu time = 00:00:17, elapsed time = 00:00:17, memory = 1014.63 (MB), peak = 1014.63 (MB)
[INFO DRT-0187] Start routing data preparation.
[INFO DRT-0267] cpu time = 00:00:00, elapsed time = 00:00:00, memory = 1014.63 (MB), peak = 1014.63 (MB)
[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
    Completing 10% with 0 violations.
    elapsed time = 00:01:24, memory = 1015.13 (MB).
    Completing 20% with 0 violations.
    elapsed time = 00:03:12, memory = 1128.25 (MB).
Error: pnr.tcl, 126 vector::_M_range_check: __n (which is 9) >= this->size() (which is 9)
openroad>
```

titan73, 2025-06-30:
> Needs a test case

I've uploaded a test case full_test_case_2025-06-30.zip on PII server.

titan73, 2025-07-10:
Actually the issue is still here. It's just happening in next iteration (1).

maliberty, 2025-07-15:
catch throw

titan73, 2025-07-15:
Thanks. I reach a catchpoint but it's not the correct one. After continuing, I get the assert but without reaching a catchpoint. Is that normal?

```
[INFO DRT-0195] Start 0th optimization iteration.
    Completing 10% with 0 violations.
    elapsed time = 00:01:32, memory = 1014.21 (MB).

Thread 1 "openroad" hit Catchpoint 1 (exception thrown), 0x00007ffff36ae4a1 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff36ae4a1 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff36a54cd in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00005555583490a9 in std::vector<std::unique_ptr<drt::frLayer, std::default_delete<drt::frLayer> >, std::allocator<std::unique_ptr<drt::frLayer, std::default_delete<drt::frLayer> > > >::at(unsigned long) const [clone .isra.0] ()
#3  0x00005555583526ba in drt::FlexGCWorker::Impl::patchMetalShape_minStep() ()
#4  0x000055555835dbb6 in drt::FlexGCWorker::Impl::patchMetalShape() ()
#5  0x00005555563c889d in drt::FlexGCWorker::Impl::main() ()
#6  0x00005555584d02d0 in drt::FlexDRWorker::route_queue_main(std::queue<drt::FlexDRWorker::RouteQueueEntry, std::deque<drt::FlexDRWorker::RouteQueueEntry, std::allocator<drt::FlexDRWorker::RouteQueueEntry> > >&) ()
#7  0x00005555584d1cf2 in drt::FlexDRWorker::route_queue() ()
#8  0x00005555563cc7b6 in drt::FlexDRWorker::main(drt::frDesign*) ()
#9  0x0000555558493144 in drt::FlexDR::processWorkersBatch(std::vector<std::unique_ptr<drt::FlexDRWorker, std::default_delete<drt::FlexDRWorker> >, std::allocator<std::unique_ptr<drt::FlexDRWorker, std::default_delete<drt::FlexDRWorker> > > >&, drt::FlexDR::IterationProgress&) [clone ._omp_fn.0] ()
#10 0x00007ffff7da2a16 in GOMP_parallel () from /lib/x86_64-linux-gnu/libgomp.so.1
#11 0x000055555847f5d4 in drt::FlexDR::processWorkersBatch(std::vector<std::unique_ptr<drt::FlexDRWorker, std::default_delete<drt::FlexDRWorker> >, std::allocator<std::unique_ptr<drt::FlexDRWorker, std::default_delete<drt::FlexDRWorker> > > >&, drt::FlexDR::IterationProgress&) ()
#12 0x000055555848f590 in drt::FlexDR::optimizationFlow(drt::FlexDR::SearchRepairArgs const&, drt::FlexDR::IterationProgress&) ()
#13 0x000055555848f8b9 in drt::FlexDR::searchRepair(drt::FlexDR::SearchRepairArgs const&) ()
#14 0x00005555563cbd3f in drt::FlexDR::main() ()
#15 0x00005555563c93ee in drt::TritonRoute::main() ()
#16 0x0000555556de865d in detailed_route_cmd(char const*, char const*, char const*, char const*, char const*, bool, int, char const*, char const*, int, double, char const*, char const*, int, bool, bool, bool, int, bool, char const*, int) ()
#17 0x0000555556debdfa in _wrap_detailed_route_cmd ()
#18 0x00007ffff7e36d32 in TclNRRunCallbacks () from /lib/x86_64-linux-gnu/libtcl8.6.so
#19 0x00007ffff7e38028 in ?? () from /lib/x86_64-linux-gnu/libtcl8.6.so
#20 0x00007ffff7e37aa7 in Tcl_EvalEx () from /lib/x86_64-linux-gnu/libtcl8.6.so
#21 0x00007ffff7e38c9a in Tcl_Eval () from /lib/x86_64-linux-gnu/libtcl8.6.so
#22 0x00005555566a0098 in sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) ()
#23 0x00005555563ea543 in tclAppInit(int&, char**, char const*, Tcl_Interp*) [clone .constprop.0] ()
#24 0x00007ffff7eeab19 in Tcl_MainEx () from /lib/x86_64-linux-gnu/libtcl8.6.so
#25 0x00005555563a12fe in main ()
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) n
Program not restarted.
(gdb) cont
Continuing.
    Completing 20% with 0 violations.
    elapsed time = 00:09:14, memory = 1127.36 (MB).
Error: pnr.tcl, 126 vector::_M_range_check: __n (which is 9) >= this->size() (which is 9)
openroad> [New Thread 0x7fffee9f9640 (LWP 5379)]
```

maliberty, 2025-07-15:
Your stack looks like you are at the point of the assertion.

titan73, 2025-07-17:
Yes indeed. The assert happens in the method:
```
// loop through violation and patch C5 enclosure minStep for GF14
void FlexGCWorker::Impl::patchMetalShape_minStep()
```
Is this method specific to GF14?

The assert comes this code when calling tech_->getLayer(lNum + 1):
```
tech_->getLayer(lNum + 1)
    auto lNum = marker->getLayerNum();
    auto layer = tech_->getLayer(lNum);
    if (!layer->hasVia2ViaMinStepViol()
        && !tech_->getLayer(lNum - 1)->hasLef58MaxSpacingConstraints()
        && !tech_->getLayer(lNum + 1)->hasLef58MaxSpacingConstraints()) {
      continue;
    }
```
The violation is on metal4 which is the last metal layer in the techno I'm using. The assert happens because there is no layer above. I guess we should check that with:
`
        && (lNum > tech_->getLayers().size() || !tech_->getLayer(lNum + 1)->hasLef58MaxSpacingConstraints())) {
`
I can send a PR for that.

It's surprising there is a violation on that layer as the rooting is limited to metal3 with:
`set_routing_layers -signal metal1-metal3 -clock metal1-metal3`

maliberty, 2025-07-17:
A PR would be good.  This looks to be patching a specific sort of min step related to vias at a transition layer.  

I don't know why we would see a violation outside the routing layer range.  @osamahammad21 any ideas?

maliberty, 2025-07-17:
The CI uses 18.1.3

titan73, 2025-07-18:
I have another case in another techno where I use:
`set_routing_layers -signal metal2-metal4B -clock metal2-metal4B`
metal5 is not used for routing which is expected but metal1 is used which shouldn't happen:
```
[INFO DRT-0199]   Number of violations = 56076.
Viol/Layer      metal1     v1 metal2     v2 metal3    v3B me..4B
Cut Spacing          0      0      0      4      0      4      0
Metal Spacing       21      0   5832      0   4104      0   3988
Min Hole             0      0      1      0      0      0      2
Recheck              0      0      3      0      7      0      1
Short               42      2  24109     23  15311      2   1632
SpacingRange       835      0    109      0     44      0      0
```

Is this a bug?

maliberty, 2025-07-18:
You still have to put vias to m1.  If you see actual routing that would be an issue.

titan73, 2025-07-19:
Yes there is routing in metal1:
[image]

I'll create an issue then.

osamahammad21, 2025-07-20:
Hi @titan73 
Sorry for the late reply. Let me first clarify why there is routing on metal4. `set_routing_layers` is actually a **grt**(global route) command which specifies which layers should be used the global router. That doesn't enforce any layer range for **drt**(detailed route). If you want to limit the detailed router by metal4, you should specify -top_routing_layer argument in the detailed_route command.
Once I have more information on the test case, I will add more updates.

osamahammad21, 2025-07-20:
> [@osamahammad21](https://github.com/osamahammad21) we should have the one command control both rather than have two different controls. I can't see when you would want these to differ.

DRT is encouraged to follow GRT's guides but not enforced. So the layer range set at the GRT stage is more of a soft requirement to reduce overall congestion on higher layers or maybe avoid unwanted parasitics. With this in mind, DRT could still use layers out of that layer range during routing to skip local congestion or hard violations. However, the layer range set at the detailed route stage is a hard requirement that cannot be violated. If we make one command for both, we lose this distinction. I don't know of any practical test case that could benefit from this.

@maliberty If you confirm we don't need this separation, I could go ahead with a PR that makes that command a global OpenROAD command for both GRT and DRT.

maliberty, 2025-07-20:
I think that flexibility unlikely to be used but if we want to preserve it then it should be opt-in.  If -top_routing_layer  is unspecified then it should default to what set_routing_layers used.  -bottom_routing_layer should be the same.

@gadfort any opinion?

gadfort, 2025-07-20:
@maliberty I can't think of a case there I would want to artificially provide the wrong routing layers to GRT and the correct ones to DRT. One command would simplify this I think and avoid issues related to needing to call drt with the same layers as grt (or risk getting errors about guides outside the layers).