The-OpenROAD-Project/OpenROAD #7717: [ERROR RSZ-0501] buffering pin _1339229_/Y failed: area recovery cannot reproduce solution

oharboe, 2025-07-04:
### Describe the bug

untar and run https://drive.google.com/file/d/10V2JTn74X8GLBwMMSPtawusSMFXOCwyV/view?usp=sharing

Wait 1h40

```
[INFO RSZ-0039] Resized 332297 instances.
[INFO RSZ-0038] Inserted 12193 buffers in 10894 nets.
   Iter   |    Area   | Removed | Inserted |   Pins
          |           | Buffers | Buffers  | Remaining
-------------------------------------------------------
        0 |     +0.0% |       0 |        0 |    949921
    94900 |     -0.2% |   14874 |    15116 |    855021
   189800 |     -0.4% |   31029 |    32216 |    760121
[ERROR RSZ-0501] buffering pin _1339229_/Y failed: area recovery cannot reproduce solution
Error: global_place.tcl, 51 RSZ-0501
Command exited with non-zero status 1
Elapsed time: 1:39:53[h:]min:sec. CPU time: user 24220.01 sys 6160.32 (506%). Peak memory: 14717356KB.
make[1]: *** [external/bazel-orfs++orfs_repositories+docker_orfs/OpenROAD-flow-scripts/flow/Makefile:469: do-3_3_place_gp] Error 1
make: *** [external/bazel-orfs++orfs_repositories+docker_orfs/OpenROAD-flow-scripts/flow/Makefile:489: do-place] Error 2
```

### Expected Behavior

No error

### Environment

```shell
docker.io/openroad/orfs:v3.0-3327-g73e5053d
```

### To Reproduce

See above

### Relevant log output

```shell

```

oharboe, 2025-07-04:
Making an attempt at deltaDebug.py to bring down running times:

```
$ openroad -python ~/OpenROAD-flow-scripts/tools/OpenROAD/etc/deltaDebug.py --use_stdout --exit_early_on_error --error_string RSZ-0501 --base_db_path results/asap7/BoomTile/base/3_2_place_iop.odb --step ./run-me-BoomTile-asap7-base.sh --persistence=4
OpenROAD v2.0-22612-ga090ebaa03 
[deleted]
Performing a step with the original input file to calculate timeout.
[running]
```

maliberty, 2025-07-04:
Similar issues were fixed with https://github.com/The-OpenROAD-Project/OpenROAD/pull/7648.  Are you using that?

oharboe, 2025-07-05:
Fixed!