The-OpenROAD-Project/OpenROAD #7746: 2_4_floorplan_pdn has different hash locally and in bazel CI

oharboe, 2025-07-08:
### Describe the bug

To reproduce, untar and run [pdn_MockArray_asap7_base_2025-07-08_10-18.tar.gz](https://github.com/user-attachments/files/21117739/pdn_MockArray_asap7_base_2025-07-08_10-18.tar.gz)

Then to display hash:

```
$ sha1sum test/orfs/mock-array/results/asap7/MockArray/base/2_4_floorplan_pdn.odb 
3e006640e8c1afb7923e5893c47680c2183b3625  test/orfs/mock-array/results/asap7/MockArray/base/2_4_floorplan_pdn.odb
```

```
$ bazelisk run test/orfs/mock-array:MockArray_floorplan_deps /tmp/pdn
$ /tmp/pdn/make do-floorplan

```

Local:

```
$ bazelisk test test/orsf/mock-array:MockArray_test
[deleted]
$ grep -h -r "Elapsed/s" -A1 bazel-bin/test/orfs/mock-array/logs/asap7/MockArray/base/*.log | grep -v "Elapsed/s" | grep -v "^--$"
2_1_floorplan                      1            208 40097d51797c6c473b09
2_2_floorplan_macro                3            188 e3184139724d8d72b5bc
2_3_floorplan_tapcell              1            166 717531cdace8f3c5254c
2_4_floorplan_pdn                 63           4102 3e006640e8c1afb7923e
3_1_place_gp_skip_io               5            228 0115c95b13660ab5b33a
3_2_place_iop                      1            184 6611bceadd4dda49af52
3_3_place_gp                      17            843 44b0ce0abd8e3e4e10e5
3_4_place_resized                  3            243 44b0ce0abd8e3e4e10e5
3_5_place_dp                       5            889 e0f0f3fa9b743840b9f5
4_1_cts                            6            902 df5920d204fe4f71395f
5_1_grt                           14           1656 cf605726892fcd8e5e37
5_2_route                        447           4717 79a1a97a3bcf860c8930
5_3_fillcell                       2            873 b4b5eba5f05bdd75f29c
6_report                           5            324                  N/A
```

From https://github.com/The-OpenROAD-Project/OpenROAD/pull/7722 and the output on the CI https://jenkins.openroad.tools/job/OpenROAD-Public/job/PR-7722-head/7/pipeline-overview/?selected-node=59

```
[2025-07-08T07:22:31.008Z] Elapsed time: 0:02.53[h:]min:sec. CPU time: user 2.38 sys 0.14 (99%). Peak memory: 171096KB.
[2025-07-08T07:22:31.008Z] Log                        Elapsed/s Peak Memory/MB  sha1sum .odb [0:20)
[2025-07-08T07:22:31.008Z] 2_3_floorplan_tapcell              2            167 717531cdace8f3c5254c
[2025-07-08T07:22:31.008Z] external/bazel-orfs++orfs_repositories+docker_orfs/OpenROAD-flow-scripts/flow/scripts/flow.sh 2_4_floorplan_pdn pdn
[2025-07-08T07:22:31.008Z] Running pdn.tcl, stage 2_4_floorplan_pdn
[2025-07-08T07:22:31.008Z] source external/bazel-orfs++orfs_repositories+docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/liberty_suppressions.tcl
[2025-07-08T07:22:31.009Z] suppress_message STA 1212
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: top
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_0_0
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_0_1
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_0_2
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_0_3
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_0_4
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_0_5
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_0_6
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_0_7
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_1_0
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_1_1
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_1_2
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_1_3
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_1_4
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_1_5
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_1_6
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_1_7
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_2_0
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_2_1
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_2_2
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_2_3
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_2_4
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_2_5
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_2_6
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_2_7
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_3_0
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_3_1
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_3_2
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_3_3
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_3_4
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_3_5
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_3_6
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_3_7
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_4_0
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_4_1
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_4_2
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_4_3
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_4_4
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_4_5
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_4_6
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_4_7
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_5_0
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_5_1
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_5_2
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_5_3
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_5_4
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_5_5
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_5_6
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_5_7
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_6_0
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_6_1
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_6_2
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_6_3
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_6_4
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_6_5
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_6_6
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_6_7
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_7_0
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_7_1
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_7_2
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_7_3
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_7_4
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_7_5
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_7_6
[2025-07-08T07:22:31.009Z] [INFO PDN-0001] Inserting grid: ElementGrid - ces_7_7
[2025-07-08T07:22:31.009Z] Elapsed time: 3:45.63[h:]min:sec. CPU time: user 218.85 sys 6.70 (99%). Peak memory: 4198992KB.
[2025-07-08T07:22:31.009Z] Log                        Elapsed/s Peak Memory/MB  sha1sum .odb [0:20)
[2025-07-08T07:22:31.009Z] 2_4_floorplan_pdn                225           4100 4d53f67594daf3078698
```

### Expected Behavior

Same result always

### Environment

```shell
v2.0-22706-g9c4e436fd2
```

### To Reproduce

See above

### Relevant log output

```shell

```

maliberty, 2025-07-09:
I was able to reproduce this with OR commit 10ae1161a7.  @gadfort do you want to look at this?

maliberty, 2025-07-09:
Note that if I write a DEF from each db the results are identical.  It must be the order of elements that varies in the odb.