The-OpenROAD-Project/OpenROAD #7752: CTS crash once in a blue moon

oharboe, 2025-07-08:
### Describe the bug

I've seen this before, this is the most information I have about it, doesn't always happen. If it happens, reproducible in minutes.

Untar and run and if the stars align, you'll get a crash: https://drive.google.com/file/d/1RiFG9ZJx85ZUshXdAst8IKwHfKU83DIG/view?usp=sharing

```
[INFO CTS-0098] Clock net "clock_regs"
[INFO CTS-0099]  Sinks 121655
[INFO CTS-0100]  Leaf buffers 3558
[INFO CTS-0101]  Average sink wire length 932.41 um
[INFO CTS-0102]  Path depth 23 - 24
[INFO CTS-0207]  Leaf load cells 3246
[INFO RSZ-0047] Found 61 long wires.
[INFO RSZ-0048] Inserted 155 buffers in 61 nets.
malloc(): corrupted top size
Signal 6 received
Stack trace:
```

```
futex_wait (private=0, expected=2, futex_word=0x72d37121ac80 <main_arena>)
    at ../sysdeps/nptl/futex-internal.h:146
warning: 146 ../sysdeps/nptl/futex-internal.h: No such file or directory
(gdb) bt
#0  futex_wait (private=0, expected=2, futex_word=0x72d37121ac80 <main_arena>)
    at ../sysdeps/nptl/futex-internal.h:146
#1  __GI___lll_lock_wait_private (
    futex=futex@entry=0x72d37121ac80 <main_arena>) at ./nptl/lowlevellock.c:34
#2  0x000072d3710a53c8 in __GI___libc_malloc (bytes=1665)
    at ./malloc/malloc.c:3327
#3  0x000072d3714ae98c in operator new (sz=1665)
    at ../../../../src/libstdc++-v3/libsupc++/new_op.cc:50
#4  0x00005ca2edf146d9 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::reserve(unsigned long) ()
#5  0x00005ca2edf13ef9 in handler(int) ()
#6  <signal handler called>
#7  __pthread_kill_implementation (no_tid=0, signo=6, threadid=126252488169856)
    at ./nptl/pthread_kill.c:44
#8  __pthread_kill_internal (signo=6, threadid=126252488169856)
    at ./nptl/pthread_kill.c:78
#9  __GI___pthread_kill (threadid=126252488169856, signo=signo@entry=6)
    at ./nptl/pthread_kill.c:89
#10 0x000072d371042476 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/posix/raise.c:26
#11 0x000072d3710287f3 in __GI_abort () at ./stdlib/abort.c:79
#12 0x000072d371089677 in __libc_message (action=action@entry=do_abort,
    fmt=fmt@entry=0x72d3711dbb77 "%s\n") at ../sysdeps/posix/libc_fatal.c:156
--Type <RET> for more, q to quit, c to continue without paging--
#13 0x000072d3710a0cfc in malloc_printerr (
    str=str@entry=0x72d3711d97ec "malloc(): corrupted top size")
    at ./malloc/malloc.c:5664
#14 0x000072d3710a46f2 in _int_malloc (
    av=av@entry=0x72d37121ac80 <main_arena>, bytes=bytes@entry=1436648)
    at ./malloc/malloc.c:4373
#15 0x000072d3710a5139 in __GI___libc_malloc (bytes=1436648)
    at ./malloc/malloc.c:3329
#16 0x000072d3714ae98c in operator new (sz=1436648)
    at ../../../../src/libstdc++-v3/libsupc++/new_op.cc:50
#17 0x00005ca2ee288686 in sta::BfsIterator::visitParallel(int, sta::VertexVisitor*) ()
#18 0x00005ca2ee2d5cf4 in sta::Search::findClkArrivals() ()
#19 0x00005ca2efcfdc06 in cts::TritonCTS::balanceMacroRegisterLatencies() ()
#20 0x00005ca2efcfe44c in cts::TritonCTS::runTritonCts() ()
#21 0x00005ca2ee6d90b1 in _wrap_run_triton_cts ()
#22 0x000072d3760ddd32 in TclNRRunCallbacks (
    interp=interp@entry=0x5ca2feab65a0, result=0, rootPtr=0x0)
    at ./generic/tclBasic.c:4494
#23 0x000072d3760dd705 in Tcl_EvalObjv (interp=interp@entry=0x5ca2feab65a0,
    objc=objc@entry=4, objv=objv@entry=0x5ca2feac3ea0,
    flags=flags@entry=2097168) at ./generic/tclBasic.c:4217
#24 0x000072d3760df028 in TclEvalEx (interp=interp@entry=0x5ca2feab65a0,
    script=0x5ca2fee5b9b0 "sta::include_file external/bazel-orfs++orfs_repositories+docker_orfs/OpenROAD-flow-scripts/flow/scripts/cts.tcl 0 0",
    numBytes=<optimized out>, numBytes@entry=-1, flags=flags@entry=0,
    line=line@entry=1, clNextOuter=clNextOuter@entry=0x0,
    outerScript=0x5ca2fee5b9b0 "sta::include_file external/bazel-orfs++orfs_repositories+docker_orfs/OpenROAD-flow-scripts/flow/scripts/cts.tcl 0 0")
    at ./generic/tclBasic.c:5363
#25 0x000072d3760deaa7 in Tcl_EvalEx (interp=interp@entry=0x5ca2feab65a0,
    script=<optimized out>, numBytes=numBytes@entry=-1, flags=flags@entry=0)
    at ./generic/tclBasic.c:5028
#26 0x000072d3760dfc9a in Tcl_Eval (interp=0x5ca2feab65a0,
    script=<optimized out>) at ./generic/tclBasic.c:5957
#27 0x00005ca2ee174167 in sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) ()
#28 0x00005ca2edf135f3 in tclAppInit(int&, char**, char const*, Tcl_Interp*) [clone .constprop.0] ()
#29 0x000072d376191b19 in Tcl_MainEx (argc=<optimized out>,
    argv=0x7ffc745cc2d0,
    appInitProc=0x5ca2edf13c30 <ord::tclAppInit(Tcl_Interp*)>,
    interp=0x5ca2feab65a0) at ./generic/tclMain.c:371
#30 0x00005ca2edeca02e in main ()
(gdb) 
```

### Expected Behavior

No crash.

### Environment

```shell
v2.0-22712-g2d97822cba
```

### To Reproduce

See above.

### Relevant log output

```shell

```

oharboe, 2025-07-08:
Tried with 2d97822cba0583ea21dba890c6713f634ba28f26

    bazelisk build -c opt :openroad
    export PATH=$(pwd)/bazel-bin:$PATH

Modify run*.sh script to exit with non-zero exit code on error, then run:

    { for i in {1..100}; do ./run*.sh && echo 0 || echo 1; done; } | sort | uniq -c

Output:

SimLogic, 2025-07-09:
Caused by glibc memory dead-lock, when it still inside critical-area, while async signal happens.

glibc is not async-signal safe....

oharboe, 2025-07-09:
> Any idea what the async signal might be or how to remedy it?

ChatGPT says: Tools like glibc's malloc, ASan, or Valgrind may call abort() if they detect misuse, causing signal 6.

oharboe, 2025-07-09:
I ran this example with valgrind and it finds a ton of things it objects to, various categories, such as:

```
==547181== Mismatched free() / delete / delete []
==547181==    at 0x5B878BF: free (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==547181==    by 0x3A8D15D: odb::dbIStream& odb::operator>><float, 4096u, 12u>(odb::dbIStream&, odb::dbPagedVector<float, 4096u, 12u>&) (in /home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/execroot/_main/bazel-out/k8-opt/bin/openroad)
==547181==    by 0x3A77F7E: odb::operator>>(odb::dbIStream&, odb::_dbBlock&) (in 
```

SimLogic, 2025-07-09:
Based upon stack-trace, the root-cause is due to #6 (which is async-signal handler).
But, yes, the trigger is coming from corrupted memory, which tries to print message during malloc/free, while async-signal jumping into...

You might check this for more details, about solution.
https://docs.oracle.com/cd/E19455-01/806-5257/gen-26/index.html

maliberty, 2025-07-09:
valgrind has been running all night but no issues have been flagged so far in my run.

oharboe, 2025-07-09:
> valgrind has been running all night but no issues have been flagged so far in my run.

strange. That answers my unspoken question: we expect no valgrind errors here.

I think I used `bazelisk build -c opt :openroad` and not cmake build.

maliberty, 2025-07-09:
My run is just -g

oharboe, 2025-07-09:
> rules_cc::cc::runfiles is not part of openroad afaik.  I'm guessing you are running it on bazel not OR

no, runfiles is linked w OR to find e.g. tcl runfiles if you recall.

The original crash in this issue was optimized cmake OR to be clear.

oharboe, 2025-07-09:
> [@arthurjolo](https://github.com/arthurjolo) is running it in a loop to see if we can see it happen. I assume he is using a cmake build. If it doesn't then I agree we can close it.

I had no luck with that.

One thing this PR does do is to illustrate that valgrind + bazel needs more work... 😺

oharboe, 2025-07-10:
> I left the testcase running throw the night in a loop. I didn't get any crash, it was still running this morning. I used cmake to build OR.

Same here.

We need a better reproduction procedure or it is too costly to pursue. Closing.

closed 2025-07-10 by oharboe