The-OpenROAD-Project/OpenROAD #7779: Illegal net/pin combination with wrapped operators and hierarchical linking

precisionmoon, 2025-07-10:
### Describe the bug

Wrapped operators with link_design -hier led to the following error on 3.3 place_gp step on multiple designs:
[ERROR ORD-2024] Illegal net/pin combination. Modnets can only be hooked to iterm, bterm, moditerm, modbterm, dbNets can only be hooked to iterm, bterm

asap7 jpeg
asap7 swerv_wrapper
nangate45 bp_be
nangate45 bp_fe
nangate45 bp_multi
nangage45 mempool

### Expected Behavior

Mod net creation may be faulty.

### Environment

```shell
Error: Access to the public repository 'The-OpenROAD-Project/OpenROAD' is restricted.
Please use the private repositories for your work.
```

### To Reproduce

To reproduce, run ORFS with SYNTH_WRAPPED_OPERATORS=1 and modify scripts/load.tcl to enable link -hier and read_db -hier.

### Relevant log output

```shell

```

maliberty, 2025-07-11:
I'm planning to have Jaehyun look at hierarchy support.  If you think it would be quick for you I'm open to splitting it.

povik, 2025-07-11:
I'd prefer if I could switch over to other pending tasks.

maliberty, 2025-07-11:
That's fine