The-OpenROAD-Project/OpenROAD #7783: Rebuffer::fullyRebuffer fatals with wrapped operator synthesis

precisionmoon, 2025-07-10:
### Describe the bug

Wrapped operator synthesis led to a fatal during GP resizer:

Iteration | Overflow |     HPWL (um) |  HPWL(%) |   Penalty | Group
---------------------------------------------------------------
      350 |   0.2507 |  5.427641e+07 |  +17.62% |  4.78e-06 |      
      360 |   0.2333 |  5.319687e+07 |   -1.99% |  7.05e-06 |      
      370 |   0.2075 |  5.297075e+07 |   -0.43% |  1.04e-05 |      
[INFO GPL-0100] Timing-driven iteration 4/5, virtual: false.
[INFO GPL-0101]    Iter: 372, overflow: 0.204, keep resizer changes at: 1, HPWL: 52935177
Iteration |   Area    | Resized | Buffers | Nets repaired | Remaining
---------------------------------------------------------------------
        0 |     +0.0% |       0 |       0 |             0 |     11350
    final |     +1.5% |    2247 |       0 |             0 |         0
---------------------------------------------------------------------
[INFO RSZ-0039] Resized 2247 instances.
   Iter   |    Area   | Removed | Inserted |   Pins
          |           | Buffers | Buffers  | Remaining
-------------------------------------------------------
        0 |     +0.0% |       0 |        0 |      9161
      910 |     -0.2% |      89 |       80 |      8251

double free or corruption (!prev)
Signal 6 received
Program received signal SIGSEGV, Segmentation fault.
0x00005555566dd2b2 in sta::Vertex::setBfsInQueue(sta::BfsIndex, bool) ()
(gdb) where
#0  0x00005555566dd2b2 in sta::Vertex::setBfsInQueue(sta::BfsIndex, bool) ()
#1  0x000055555679a481 in sta::BfsIterator::next() ()
#2  0x00005555567e82dd in sta::Search::levelsChangedBefore() ()
#3  0x00005555567a8a1e in sta::Levelize::levelize() ()
#4  0x0000555558889133 in rsz::Rebuffer::fullyRebuffer(sta::Pin*) ()
#5  0x00005555587f8ccd in rsz::Resizer::findResizeSlacks(bool) ()
#6  0x0000555556516ff9 in gpl::TimingBase::executeTimingDriven(bool) ()
#7  0x00005555564c1828 in gpl::NesterovPlace::doNesterovPlace(int) ()
#8  0x00005555564b5a22 in gpl::Replace::doNesterovPlace(int, int) ()
#9  0x000055555655a3eb in _wrap_replace_nesterov_place_cmd ()
#10 0x00007ffff7e3a2f9 in TclNRRunCallbacks (interp=interp@entry=0x55555af0da20, result=0, rootPtr=0x0)
    at /usr/src/tcl8.6-8.6.14+dfsg-1build1/generic/tclBasic.c:4539

### Expected Behavior

No fatal.

### Environment

```shell
Error: Access to the public repository 'The-OpenROAD-Project/OpenROAD' is restricted.
Please use the private repositories for your work.
```

### To Reproduce

Run ORFS with SYNTH_WRAPPED_OPERATORS=1 and modify scripts/load.tcl to enable link -hier and read_db -hier.

### Relevant log output

```shell

```

povik, 2025-07-11:
@precisionmoon Which design this is?

precisionmoon, 2025-07-11:
Sorry, this is in asap7 / riscv32i.  The fatal happens in 3_3_place_gp step.

precisionmoon, 2025-07-11:
These designs show similar fatals:

gf12 ariane
gf12 ca53_cpu