The-OpenROAD-Project/OpenROAD #7812: Have gui and non-gui builds in bazel similar to cmake

maliberty, 2025-07-15:
### Description

In cmake we build OR both with and without the GUI.  Now that we have the GUI working in bazel I think we should do the same.  The non-gui build does not need to run any tests.  See https://github.com/The-OpenROAD-Project/OpenROAD/pull/7725 for details on the GUI.

oharboe, 2025-07-15:
Suggestion: just add a test that smoketests both and create a openroad-gui target, no change to CI required.

oharboe, 2025-07-15:
@QuantamHD Thoughts on adding a openroad-gui target alongside openroad and adding a simple test that tests both and thus `bazelisk test ...` in CI discover and test both targets?

If we have openroad and openroad-gui targets, that would require changes to bazel builds.

If we need to build GUI and non-GUI builds in the future, perhaps these targets should be renamed to openroad (with GUI) and openroad-cli (no gui)

oharboe, 2025-07-15:
The point here is: keep the CI interface to bazel simple and stable, solve these issues in bazel.

oharboe, 2025-07-15:
> [@oharboe](https://github.com/oharboe) right now there are no GUI specific tests, my request here was just to ensure we build with the GUI enabled to make sure that works.

Yes: and that can be achieved without changing the interface between CI and the Bazelisk build. If we want to build with and without and do a smoketest, then we just add a few tests for either.

Since Bazel builds are not part of deployment yet, we could also just switch to building w/gui by default and have non-gui be a config option. That too would leave the CI/Bazel interface unchanged(i.e. no changes to github actions).

QuantamHD, 2025-07-15:
Can't we just add another Jenkins action that builds with that config?

oharboe, 2025-07-15:
> Can't we just add another Jenkins action that builds with that config?

You mean run all tests?

I think we should keep the CI/Bazel interface stable(no changes to CI) and just do what we want to do in Bazel.

We messed about with the CI/Bazel interface in our project and eventually settled on that keeping the local and CI testing interface stable and doing what was needed in Bazel as an implementation detail, caused the least disruption.

maliberty, 2025-07-15:
In cmake all we test is that you can build without the GUI without a compiler error.  All functional testing is done in the build with the GUI.