The-OpenROAD-Project/OpenROAD #7816: ORD Terminal: Pressing TAB to list files & dirs outputs an error

VoltXd, 2025-07-15:
### Describe the bug

When I run openroad in the terminal, pressing TAB after typing a command name (e.g. read_db) outputs the following error:
```
openroad> read_db [ERROR ORD-0007] ---- does not exist.
[ERROR STA-0568] read_db requires one or two positional arguments.
```
The same kind of error happens for other commands, I tested
```
openroad> gui::show [ERROR GUI-0070] invalid command name "----"
``` 
and 
```
openroad> write_lef [ERROR STA-0565] write_lef requires one positional argument.
```

If I press TAB again, here is the total output:
```
openroad> read_db [ERROR ORD-0007] ---- does not exist.
[ERROR STA-0568] read_db requires one or two positional arguments.
[ERROR ORD-0007] ---- does not exist.
[ERROR STA-0568] read_db requires one or two positional arguments.

bin/                   compile_commands.json  Makefile
CMakeCache.txt         CTestTestfile.cmake    src/
CMakeFiles/            _deps/                 test/
cmake_install.cmake    lib/                   third-party/
openroad> read_db
```

### Expected Behavior

Only the files and directories present in the current directory should be output, with my previous example:
```
openroad> read_db

bin/                   compile_commands.json  Makefile
CMakeCache.txt         CTestTestfile.cmake    src/
CMakeFiles/            _deps/                 test/
cmake_install.cmake    lib/                   third-party/
openroad> read_db
```

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.8.0-60-generic
os: Ubuntu 24.04.2 LTS (Noble Numbat)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:75 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 13.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: ccda6ca36763875abd64ab1e2724ddbb06cd81ef
-- System name: Linux
-- Compiler: GNU 13.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.3.0
[... 130 lines cut ...]
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found OpenMP: TRUE (found version "4.5")  
-- GUI is enabled
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Eigen3: /home/peyronnet/.local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.IACb0grniU
```

### To Reproduce

Write any command with a space in the OpenROAD terminal, then press TAB.

### Relevant log output

```shell

```

VoltXd, 2025-07-16:
I agree with that, however in a normal terminal when you press TAB you can see the content of the current directory, just to let you know what filename you can type. Since OpenROAD does this when you press TAB multiple times, I don't understand why an error message appears, as if you pressed Enter to execute the command, then TAB to display the directory content. 

Can you tell me where the code of the terminal is located ?

gadfort, 2025-07-16:
This comes from the TCL readline addon. I think the issue is that the TCL completer is running when it probably shouldn't be.
`Built-in Completer: ::tclreadline::readline builtincompleter can be used to enable or disable the default built-in completion mechanism. `