The-OpenROAD-Project/OpenROAD #7874: error post build GDSII

y20n20, 2025-07-23:
### Describe the bug

ending the make of DESIGN_CONFIG=./designs/sky130hs/ibex/config.mk with errors related to openroad app.

[complete_build_with_error_at_the_end.txt](https://github.com/user-attachments/files/21389248/complete_build_with_error_at_the_end.txt)

######################################
Cell type report:                       Count       Area
  Fill cell                             21384  156117.33
  Tap cell                               4462    7132.06
  Antenna cell                             44     140.66
  Clock buffer                            764   23809.77
  Timing Repair Buffer                    715   12620.97
  Inverter                                532    3524.47
  Clock inverter                          373    3048.15
  Sequential cell                        1932   75096.03
  Multi-Input combinational cell         9304  129746.92
  Total                                 39510  411236.35
Report metrics stage 6, finish...

==========================================================================
finish report_design_area
--------------------------------------------------------------------------
Design area 247987 u^2 60% utilization.
[WARNING GUI-0076] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
MESA: error: Failed to query drm device.
libGL error: glx: failed to create dri3 screen
libGL error: failed to load driver: iris
libGL error: failed to open /dev/dri/card0: No such file or directory
libGL error: failed to load driver: iris
Elapsed time: 1:15.93[h:]min:sec. CPU time: user 115.12 sys 8.58 (162%). Peak memory: 635912KB.
Log                        Elapsed/s Peak Memory/MB  sha1sum .odb [0:20)
6_report                          75            621                  N/A
cp ./results/sky130hs/ibex/base/5_route.sdc ./results/sky130hs/ibex/base/6_final.sdc
cp /OpenROAD-flow-scripts/flow/platforms/sky130hs/lef/sky130_fd_sc_hs.tlef ./objects/sky130hs/ibex/base/klayout_tech.lef
SC_LEF_RELATIVE_PATH="../../../../platforms/sky130hs/lef/sky130_fd_sc_hs_merged.lef"; \
OTHER_LEFS_RELATIVE_PATHS=$(echo "<lef-files>$(realpath --relative-to=./results/sky130hs/ibex/base ./objects/sky130hs/ibex/base/klayout_tech.lef)</lef-files>"); \
sed 's,<lef-files>.*</lef-files>,<lef-files>'"$SC_LEF_RELATIVE_PATH"'</lef-files>'"$OTHER_LEFS_RELATIVE_PATHS"',g' /OpenROAD-flow-scripts/flow/platforms/sky130hs/sky130hs.lyt > ./objects/sky130hs/ibex/base/klayout.lyt
sed -i 's,<map-file>.*</map-file>,<map-file>/OpenROAD-flow-scripts/flow/platforms/sky130hs/*map</map-file>,g' ./objects/sky130hs/ibex/base/klayout.lyt
(env time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' stdbuf -o L /usr/bin/klayout -zz -rd design_name=ibex_core \
        -rd in_def=./results/sky130hs/ibex/base/6_final.def \
        -rd in_files="/OpenROAD-flow-scripts/flow/platforms/sky130hs/gds/sky130_fd_sc_hs.gds  " \
        -rd seal_file="" \
        -rd out_file=./results/sky130hs/ibex/base/6_1_merged.gds \
        -rd tech_file=./objects/sky130hs/ibex/base/klayout.lyt \
        -rd layer_map= \
        -r /OpenROAD-flow-scripts/flow/util/def2stream.py) 2>&1 | tee /OpenROAD-flow-scripts/flow/logs/sky130hs/ibex/base/6_1_merge.log
[INFO] Reporting cells prior to loading DEF ...
	/OpenROAD-flow-scripts/flow/platforms/sky130hs/gds/sky130_fd_sc_hs.gds
[INFO] All LEF cells have matching GDS/OAS cells
[INFO] No orphan cells in the final layout
Elapsed time: 0:04.80[h:]min:sec. CPU time: user 4.45 sys 0.28 (98%). Peak memory: 532880KB.
cp results/sky130hs/ibex/base/6_1_merged.gds results/sky130hs/ibex/base/6_final.gds
./logs/sky130hs/ibex/base
Log                        Elapsed/s Peak Memory/MB  sha1sum .odb [0:20)
1_1_yosys_canonicalize             3            137 827da72f91f5ddb4378f
1_2_yosys                        134            158 7c4b9f4ddfdbfcd82e9d
2_1_floorplan                     23            198 88c99b4035a1a5c18ec3
2_2_floorplan_macro                3            149 88c99b4035a1a5c18ec3
2_3_floorplan_tapcell              3            143 97d61e7616b441c253e7
2_4_floorplan_pdn                  3            155 86c4ef1afa40859a3233
3_1_place_gp_skip_io              60            176 2caa01b0f24c2b9e4d3f
3_2_place_iop                      3            156 2da35b26eafd5c214afe
3_3_place_gp                     482            345 7249d2f60eff81a61502
3_4_place_resized                 35            209 43fddb4486553d84c0fc
3_5_place_dp                      38            259 3552547bea130f0a97b7
4_1_cts                           44            326 f1a51215160d646d1cbd
5_1_grt                          388            640 a0c54489f9458a7bba5b
5_2_route                       1490           1842 ef78cbfd66f20eff4343
5_3_fillcell                       4            230 f219783e3d13e4600120
[... 135 lines cut ...]
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.0.2", minimum required is "4.0")  
-- Found Boost 1.70.0 at /home/yaniv/miniconda3/lib/cmake/Boost-1.70.0
--   Requested configuration: REQUIRED
-- Found boost_headers 1.70.0 at /home/yaniv/miniconda3/lib/cmake/boost_headers-1.70.0
-- boost: 1.70.0
-- Configuring incomplete, errors occurred!
```

### To Reproduce

make DESIGN_CONFIG=./designs/sky130hd/filter_bank/config.mk

### Relevant log output

```shell

```

maliberty, 2025-07-23:
These look like system errors
```
MESA: error: Failed to query drm device.
libGL error: glx: failed to create dri3 screen
libGL error: failed to load driver: iris
libGL error: failed to open /dev/dri/card0: No such file or directory
libGL error: failed to load driver: iris
```
Are you running headless or in some other configuration where graphics are blocked?

y20n20, 2025-07-25:
Yes. thanks.