The-OpenROAD-Project/OpenROAD #7881: Support for upcoming 55 nm pdk

Andrea-Spitale, 2025-07-24:
### Description

Hi,
I've come across [this](https://wiki.f-si.org/images/a/ad/FSiC2025_Making_Open_Silicon_Design_Everywhere_final_version_present.pdf) presentation held for fsic 2025, I think this is the same team responsible for iEDA.

Are you perhaps aware of this upcoming 55 nm open source PDK? Looks like they should release it this month.
I am just curious to know if openroad would be open to support new PDKs.

Thanks!

gadfort, 2025-07-24:
There is it being released, the slide just says it will be released, but there is no link to where it may be found when they do release it.

Andrea-Spitale, 2025-07-24:
You're right. I have sent some emails to both the iEDA team, which seems to have already done some runs with that PDK, and the icsprout Company. I'll post here once I have updates.

QuantamHD, 2025-07-24:
The feeling I is that it's "open" not "open source". All the presentations mention that it will be available in ECOS which is their web based iEDA environment.

I strongly suspect that's the only place it'll be. Rather than being released for standard consumption

closed 2025-08-13 by maliberty