The-OpenROAD-Project/OpenROAD #7897: rcx: Crash when generating patterns for RC rules

titan73, 2025-07-26:
### Describe the bug

The command bench_wires crashes:
```
read_lef technology.lef
bench_wires -len 100 -all
```

### Expected Behavior

No crash

### Environment

```shell
OpenROAD v2.0-23106-g0e46398a83
```

### To Reproduce

Test case file crash_bench_wires.zip uploaded to PII sftp server.

### Relevant log output

```shell
OpenROAD v2.0-23106-g0e46398a83 
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
read_lef
[INFO ODB-0227] LEF file: technology_fixed.lef, created 13 layers, 45 vias
bench_wires
[INFO RCX-0055] Finished 180 bench measurements for pattern MET_OVER_MET
[WARNING ODB-0406] Cannot create net R6_M2oM0_W200W200_S00000S00210_3, duplicate net
Signal 11 received
Stack trace:
 0# 0x00005CE42D8E3732 in ./openroad
 1# 0x000070BDF5042520 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbNet::getBTerms() in ./openroad
 3# odb::dbNet::get1stBTerm() in ./openroad
 4# rcx::extMeasure::createNetSingleWire(char*, unsigned int, unsigned int, unsigned int, int) in ./openroad
 5# rcx::extRCModel::writeBenchWires_DB_res(rcx::extMeasure*) in ./openroad
 6# rcx::extRCModel::benchDB_WS(rcx::extMainOptions*, rcx::extMeasure*) in ./openroad
 7# rcx::extRCModel::linesOverBench(rcx::extMainOptions*) in ./openroad
 8# rcx::extMain::benchWires(rcx::extMainOptions*) in ./openroad
 9# rcx::Ext::bench_wires(BenchWiresOptions const&) in ./openroad
10# 0x00005CE42E099BDE in ./openroad
11# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
12# 0x000070BDF9C54028 in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
15# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in ./openroad
16# 0x00005CE42D8E2F33 in ./openroad
17# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
18# main in ./openroad
19# 0x000070BDF5029D90 in /lib/x86_64-linux-gnu/libc.so.6
20# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
21# _start in ./openroad
Segmentation fault (core dumped)
```

titan73, 2025-07-27:
I think I found the issue.

In `extRCModel::benchDB_WS`, there are the nested loop:
```
        for (uint jj = 0; jj < spaceTable->getCnt(); jj++) {
          double s = spaceTable->get(jj);  // layout
          for (uint kk = jj; kk < spaceTable->getCnt(); kk++) {
            double s2 = spaceTable->get(kk);
            measure->setTargetParams(w, s, 0.0, 0, 0, w2, s2);
            writeBenchWires_DB_res(measure);
            cnt++;
          }
        }
```

I've dumped the space table:
```
st[0] 0.000000
st[1] 0.210000
st[2] 0.210000
st[3] 0.410000
st[4] 0.520000
st[5] 0.620000
st[6] 0.820000
st[7] 0.930000
st[8] 1.030000
st[9] 1.230000
st[10] 1.640000
```

Indices 1 & 2 have the same value 0.21 which causes the duplicate prefix and probably the crash.

titan73, 2025-07-27:
Actually there are no duplicate values:
```
st[0] 0.00000000000000000000
st[1] 0.20999999999999999223
st[2] 0.21000000000000001998
st[3] 0.41000000000000003109
st[4] 0.52000000000000012879
st[5] 0.62000000000000010658
st[6] 0.82000000000000006217
st[7] 0.93000000000000004885
st[8] 1.03000000000000002665
st[9] 1.22999999999999998224
st[10] 1.64000000000000012434
```
But after rounding to generate the prefix they do.

maliberty, 2025-07-27:
@dimitrisfotakis any hints on this problem?

titan73, 2025-07-27:
I found the issue. The table is built just above the code. It's a rounding comparison issue.