The-OpenROAD-Project/OpenROAD #7928: Unable to repair hold after #7915

gadfort, 2025-07-31:
### Describe the bug

With the merge of #7915 simple designs are not able to repair hold violations.

Log before change:
```
[INFO RSZ-0046] Found 109 endpoints with hold violations.
Iteration | Resized | Buffers | Cloned Gates |   Area   |   WNS   |   TNS   | Endpoint
--------------------------------------------------------------------------------------
        0 |       0 |       0 |            0 |    +0.0% |  -0.410 | -31.894 | cpu/decode/op22$_DFFE_PP_/D
       10 |       0 |      12 |            0 |    +0.8% |  -0.359 | -27.022 | o_dbus_dat[11]$_DFFE_PP_/D
       20 |       0 |      23 |            0 |    +1.6% |  -0.356 | -23.451 | o_dbus_dat[6]$_DFFE_PP_/D
       30 |       0 |      34 |            0 |    +2.4% |  -0.353 | -19.902 | o_dbus_dat[10]$_DFFE_PP_/D
       40 |       0 |      44 |            0 |    +3.1% |  -0.350 | -16.094 | rf_ram_if/rdata0[7]$_SDFF_PN0_/D
       50 |       0 |      55 |            0 |    +3.9% |  -0.327 | -10.678 | cpu/gen_qerv_immdec.immdec/o_rd_addr[4]$_DFFE_PP_/D
       60 |       0 |      68 |            0 |    +4.8% |  -0.305 |  -6.511 | cpu/decode/opcode[1]$_DFFE_PP_/D
       70 |       0 |      69 |            0 |    +4.8% |  -0.296 |  -6.206 | cpu/gen_csr.csr/timer_irq_r$_DFFE_PP_/D
       80 |       0 |      75 |            0 |    +5.3% |  -0.278 |  -3.559 | cpu/alu/i_rs1[2]$_DFF_P_/D
       90 |       0 |      79 |            0 |    +5.5% |  -0.213 |  -2.336 | cpu/gen_qerv_immdec.immdec/o_rs1_addr[4]$_DFFE_PP_/D
      100 |       0 |      84 |            0 |    +5.9% |   0.017 |   0.000 | o_dbus_dat[27]$_DFFE_PP_/D
    final |       0 |      84 |            0 |    +5.9% |   0.017 |   0.000 | o_dbus_dat[27]$_DFFE_PP_/D
--------------------------------------------------------------------------------------
[INFO RSZ-0032] Inserted 84 hold buffers.
Placement Analysis
```

Log after:
```
[INFO RSZ-0046] Found 109 endpoints with hold violations.
Iteration | Resized | Buffers | Cloned Gates |   Area   |   WNS   |   TNS   | Endpoint
--------------------------------------------------------------------------------------
        0 |       0 |       0 |            0 |    +0.0% |  -0.410 | -31.894 | cpu/decode/op22$_DFFE_PP_/D
       10 |       0 |      24 |            0 |    +0.7% |  -0.359 | -29.033 | o_dbus_dat[11]$_DFFE_PP_/D
       20 |       0 |      54 |            0 |    +1.7% |  -0.356 | -26.393 | o_dbus_dat[6]$_DFFE_PP_/D
       30 |       0 |      84 |            0 |    +2.6% |  -0.353 | -23.751 | o_dbus_dat[10]$_DFFE_PP_/D
       40 |       0 |     113 |            0 |    +3.5% |  -0.350 | -21.122 | rf_ram_if/rdata0[7]$_SDFF_PN0_/D
       50 |       0 |     140 |            0 |    +4.4% |  -0.327 | -17.734 | cpu/gen_qerv_immdec.immdec/o_rd_addr[4]$_DFFE_PP_/D
       60 |       0 |     167 |            0 |    +5.2% |  -0.305 | -14.211 | cpu/decode/opcode[1]$_DFFE_PP_/D
       70 |       0 |     192 |            0 |    +6.0% |  -0.296 | -12.180 | cpu/gen_csr.csr/timer_irq_r$_DFFE_PP_/D
       80 |       0 |     221 |            0 |    +6.9% |  -0.278 |  -8.979 | cpu/alu/i_rs1[2]$_DFF_P_/D
       90 |       0 |     243 |            0 |    +7.6% |  -0.213 |  -6.514 | cpu/gen_qerv_immdec.immdec/o_rs1_addr[4]$_DFFE_PP_/D
      100 |       0 |     263 |            0 |    +8.2% |  -0.204 |  -4.239 | cpu/decode/opcode[0]$_DFFE_PP_/D
      110 |       0 |     264 |            0 |    +8.2% |  -0.204 |  -4.129 | cpu/decode/opcode[0]$_DFFE_PP_/D
      110 |       0 |     264 |            0 |    +8.2% |  -0.204 |  -4.129 | cpu/decode/opcode[0]$_DFFE_PP_/D
      120 |       0 |     286 |            0 |    +8.9% |  -0.095 |  -2.883 | o_dbus_dat[4]$_DFFE_PP_/D
      130 |       0 |     306 |            0 |    +9.5% |  -0.092 |  -1.953 | o_dbus_dat[14]$_DFFE_PP_/D
    final |       0 |     318 |            0 |    +9.9% |  -0.091 |  -1.404 | o_dbus_dat[23]$_DFFE_PP_/D
--------------------------------------------------------------------------------------
[WARNING RSZ-0066] Unable to repair all hold violations.
[INFO RSZ-0032] Inserted 318 hold buffers.
[ERROR RSZ-0060] Max buffer count reached.
```

### Expected Behavior

Properly repair hold violations with the minimum number of buffer cells and area impact.

### Environment

```shell
https://github.com/The-OpenROAD-Project/OpenROAD/commit/3f01e9f673aad5eb56ad76acd598c0bb7263274b
```

### To Reproduce

[sc_issue_qerv_job0_ihp130_sg13g2_stdcell_cts.repair_timing_0_20250731-014511.tar.gz](https://github.com/user-attachments/files/21520789/sc_issue_qerv_job0_ihp130_sg13g2_stdcell_cts.repair_timing_0_20250731-014511.tar.gz)

```
tar xvf sc_issue_qerv_job0_ihp130_sg13g2_stdcell_cts.repair_timing_0_20250731-014511.tar.gz
cd sc_issue_qerv_job0_ihp130_sg13g2_stdcell_cts.repair_timing_0_20250731-014511
./run.sh
```

### Relevant log output

```shell
[INFO RSZ-0046] Found 109 endpoints with hold violations.
Iteration | Resized | Buffers | Cloned Gates |   Area   |   WNS   |   TNS   | Endpoint
--------------------------------------------------------------------------------------
        0 |       0 |       0 |            0 |    +0.0% |  -0.410 | -31.894 | cpu/decode/op22$_DFFE_PP_/D
       10 |       0 |      24 |            0 |    +0.7% |  -0.359 | -29.033 | o_dbus_dat[11]$_DFFE_PP_/D
       20 |       0 |      54 |            0 |    +1.7% |  -0.356 | -26.393 | o_dbus_dat[6]$_DFFE_PP_/D
       30 |       0 |      84 |            0 |    +2.6% |  -0.353 | -23.751 | o_dbus_dat[10]$_DFFE_PP_/D
       40 |       0 |     113 |            0 |    +3.5% |  -0.350 | -21.122 | rf_ram_if/rdata0[7]$_SDFF_PN0_/D
       50 |       0 |     140 |            0 |    +4.4% |  -0.327 | -17.734 | cpu/gen_qerv_immdec.immdec/o_rd_addr[4]$_DFFE_PP_/D
       60 |       0 |     167 |            0 |    +5.2% |  -0.305 | -14.211 | cpu/decode/opcode[1]$_DFFE_PP_/D
       70 |       0 |     192 |            0 |    +6.0% |  -0.296 | -12.180 | cpu/gen_csr.csr/timer_irq_r$_DFFE_PP_/D
       80 |       0 |     221 |            0 |    +6.9% |  -0.278 |  -8.979 | cpu/alu/i_rs1[2]$_DFF_P_/D
       90 |       0 |     243 |            0 |    +7.6% |  -0.213 |  -6.514 | cpu/gen_qerv_immdec.immdec/o_rs1_addr[4]$_DFFE_PP_/D
      100 |       0 |     263 |            0 |    +8.2% |  -0.204 |  -4.239 | cpu/decode/opcode[0]$_DFFE_PP_/D
      110 |       0 |     264 |            0 |    +8.2% |  -0.204 |  -4.129 | cpu/decode/opcode[0]$_DFFE_PP_/D
      110 |       0 |     264 |            0 |    +8.2% |  -0.204 |  -4.129 | cpu/decode/opcode[0]$_DFFE_PP_/D
      120 |       0 |     286 |            0 |    +8.9% |  -0.095 |  -2.883 | o_dbus_dat[4]$_DFFE_PP_/D
      130 |       0 |     306 |            0 |    +9.5% |  -0.092 |  -1.953 | o_dbus_dat[14]$_DFFE_PP_/D
    final |       0 |     318 |            0 |    +9.9% |  -0.091 |  -1.404 | o_dbus_dat[23]$_DFFE_PP_/D
--------------------------------------------------------------------------------------
[WARNING RSZ-0066] Unable to repair all hold violations.
[INFO RSZ-0032] Inserted 318 hold buffers.
[ERROR RSZ-0060] Max buffer count reached.
```

maliberty, 2025-07-31:
Perhaps we eliminated delay buffers from hold fixing?

precisionmoon, 2025-07-31:
Current pruning is targeted for setup fixing.   For hold fixing, we need delay buffers and buffers with high drive resistance.  Let me see what can be done.

precisionmoon, 2025-07-31:
As I suspected, all the "good" hold buffers are filtered out.
Let me provide an option to disable buffer pruning as a short-term solution.
I'll do something different for hold buffers later.

**report_buffers -filtered**
Buffer Report:
There are 8 buffers that are not marked as dont-use, multi-voltage
special cells or clock buffers

There are 1 VT types:
VT type[index]: # buffers, ave leakage
  -      [0]: 8, 8.5e-10

There are 4 cell footprint types:
  BU    : 5 [62.50%]
  DLY1  : 1 [12.50%]
  DLY2  : 1 [12.50%]
  DLY4  : 1 [12.50%]

There are 1 cell site types:
  CoreSite [H=3780]: 8 [100.00%]
--------------------------------------------------------------------------------
Cell                                        Drive Drive    Leak   Site Cell   VT
                                            Res   Res*Cin          Ht Footpt Type
--------------------------------------------------------------------------------
sg13g2_buf_16                               164.8 3.1e-12 2.6e-09 3780 BU      -
sg13g2_buf_8                                329.1 3.1e-12 1.3e-09 3780 BU      -
sg13g2_buf_4                                655.3 2.7e-12 6.2e-10 3780 BU      -
sg13g2_buf_2                               1313.7 3.8e-12 3.4e-10 3780 BU      -
sg13g2_dlygate4sd2_1                       2605.2 4.2e-12 4.2e-10 3780 DLY2    -
sg13g2_dlygate4sd1_1                       2605.2 4.2e-12 3.2e-10 3780 DLY1    -
sg13g2_dlygate4sd3_1                       2612.1 4.2e-12 9.6e-10 3780 DLY4    -
sg13g2_buf_1                               2619.3 6.6e-12 2.0e-10 3780 BU      -

Filtered Buffer Report:
There are 5 buffers after filtering based on threshold voltage,
cell footprint, drive strength and cell site
--------------------------------------------------------------------------------
Cell                                        Drive Drive    Leak   Site Cell   VT
                                            Res   Res*Cin          Ht Footpt Type
--------------------------------------------------------------------------------
sg13g2_buf_16                               164.8 3.1e-12 2.6e-09 3780 BU      -
sg13g2_buf_8                                329.1 3.1e-12 1.3e-09 3780 BU      -
sg13g2_buf_4                                655.3 2.7e-12 6.2e-10 3780 BU      -
sg13g2_buf_2                               1313.7 3.8e-12 3.4e-10 3780 BU      -
sg13g2_buf_1                               2619.3 6.6e-12 2.0e-10 3780 BU      -
********************************************************************************