The-OpenROAD-Project/OpenROAD #7951: Investigate memory reductions for the pixel grid in dpl

maliberty, 2025-08-04:
### Describe the bug

The Pixel struct has grown rather large.  We should investigate the memory usage of a 1mm^2 area in asap7 and look for opportunities for improvement.  In particular blocked areas should be considered as much of the storage is not used.

### Expected Behavior

Efficient memory usage

### Environment

```shell
master branch
```

### To Reproduce

Generate a floorplan as described and run dpl.  Even with one instance to place the pixel grid can be constructed.

### Relevant log output

```shell

```

maliberty, 2025-08-04:
How would that help?  You need to store information for each site for not for an interval of sites.

maliberty, 2025-08-04:
You are changing an O(1) lookup to an O(log N) in the inner loop of the placer.  I expect this will have non-trivial impact.

maliberty, 2025-08-04:
Are you proposing to decompose all the fields of Pixel into a set of independent intervals?  You would have to look at each one (eg cell, group, sites, blocked_layers...)

QuantamHD, 2025-08-04:
I think I'm saying store all the non-spatial information in an interval tree. Probably as a packed struct.

QuantamHD, 2025-08-04:
> 

Typically, you store interval trees of interval trees one for each dimension. 

AI written example

```c++
#include <iostream>
#include <string>
#include <boost/icl/interval_map.hpp>

using ValueType = int; // The data associated with the 2D interval
using XInterval = boost::icl::discrete_interval<int>;
using YInterval = boost::icl::discrete_interval<int>;

// The inner map for the X dimension
using XMap = boost::icl::interval_map<int, ValueType>;

// The outer map holds Y-intervals mapping to X-maps
using XYMap = boost::icl::interval_map<int, XMap>;

int main() {
    // Our main 2D interval map
    XYMap xy_map;

    // Let's store a rectangle from (x=10, y=20) to (x=30, y=40)
    // with an associated value of 5.
    YInterval y_interval = YInterval::construct(20, 40);
    XInterval x_interval = XInterval::construct(10, 30);
    ValueType value = 5;

    // 1. Create the inner map for the X dimension
    XMap x_map;
    x_map.add({x_interval, value});

    // 2. Add the X-map to the outer Y-map over the Y-interval
    xy_map.add({y_interval, x_map});

    // Let's add another overlapping rectangle to see how they merge
    // Rectangle: (x=25, y=35) to (x=45, y=55) with value 10
    YInterval y2_interval = YInterval::construct(35, 55);
    XInterval x2_interval = XInterval::construct(25, 45);
    ValueType value2 = 10;
    
    XMap x2_map;
    x2_map.add({x2_interval, value2});
    xy_map.add({y2_interval, x2_map});

    // Print the resulting 2D map
    // The output shows how the map is partitioned into non-overlapping
    // rectangular regions, each with a combined value.
    std::cout << "2D Interval Map Contents:\n" << xy_map << std::endl;

    return 0;
}

```

> What is an example of non-spatial information?

Blocked, not blocked, this is the cell that's here. All the state information of pixel that is not the location of the site. As intervals get merged the value types can also be merged.

QuantamHD, 2025-08-05:
Maybe go with boost::flat_set instead? It's 24 bytes vs 56 bytes for the unordered_set.

QuantamHD, 2025-08-05:
Or if you suspect most of the time the answer is 0. Maybe just give nodes a pointer to set instead, and allocate the map dynamically on setting them.

maliberty, 2025-08-05:
With #7958 Pixel is 80 bytes.  Without the unordered_set it would be 24.  

One idea is to have two pointers in Pixel with the meaning
| First | Second | Meaning |
| :------- | :------: | -------: |
| null  | null  | empty  |
| non-null  | null  | First  |
| non-null  | non-null  | First & Second |
| null  | non-null  | Second is a pointer to a set|

I expect the last row is hardly ever used.  The whole thing could be encapsulated in a little class to hide the details.  That gets it to 40 bytes which isn't bad.

QuantamHD, 2025-08-05:
Well nullptr would be nullptr so at least that case would be fast.

maliberty, 2025-08-11:
The interval part is done.  The padding still needs work.