The-OpenROAD-Project/OpenROAD #8019: Add non-Logic delay column

oharboe, 2025-08-14:
### Description

I see Logic Delay, but what is non-Logic delay, i.e. routing?

[image]

### Suggested Solution

Add non-Logic delay column. Could it be called "Routing delay" perhaps?

maliberty, 2025-08-14:
It is also buffers and inverter pairs.   How would you use that information?

oharboe, 2025-08-14:
> It is also buffers and inverter pairs. How would you use that information?

It is the other side of the coin of levels of logic/logic delay. I'm trying to easily spot problematic routing/placement paths. If something is easy to do in logic, but hard to place and route, some RTL changes are usually needed. Having a column with non-logic delay allows me to easily look for troublesome paths without having to do mental arithmetic(subtracting clock latency and logic delay from arrival time)