The-OpenROAD-Project/OpenROAD #8088: openroad -gui crashes

clee215, 2025-08-21:
### Describe the bug

Here is the error encountered while opening openroad -gui on RHEL 9.6 Linux

NOTE:  command dnf install xcb-util* was done already as was a export=/pathway/to/existing/xcb/plugins/synopsys-tools  (eg export=QT_PLUGIN_PATH
/usr/cadence/SSV231/tools.lnx86/Qt/v5/64bit/plugins) to point to a possibly older xcb lib. None seem to help launch the gui.

It could be a RHEL 9 issue. openroad -gui works on a RHEL 9.5 ( about 3/18/25 last updated on hardware that will not update beyond ) but does not work on a fully updated RHEL 9.6

openroad> gui::show
[WARNING GUI-0076] Could not find the Qt platform plugin "xcb" in ""
[ERROR GUI-0077] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

terminate called after throwing an instance of 'std::runtime_error'
  what():  GUI-0077
Signal 6 received
Stack trace:
 0# 0x0000000000FD04F1 in openroad
 1# 0x00007FD9B6A3EBF0 in /lib64/libc.so.6
 2# 0x00007FD9B6A8BEDC in /lib64/libc.so.6
 3# raise in /lib64/libc.so.6
 4# abort in /lib64/libc.so.6
 5# 0x00007FD9B6EA5929 in /synopsys/current/customcompiler/linux64/lib/libstdc++.so.6
 6# 0x00007FD9B6EB150A in /synopsys/current/customcompiler/linux64/lib/libstdc++.so.6
 7# 0x00007FD9B6EB0579 in /synopsys/current/customcompiler/linux64/lib/libstdc++.so.6
 8# __gxx_personality_v0 in /synopsys/current/customcompiler/linux64/lib/libstdc++.so.6
 9# 0x00007FD9BB6CD391 in /synopsys/current/customcompiler/linux64/lib/libgcc_s.so.1
10# _Unwind_Resume in /synopsys/current/customcompiler/linux64/lib/libgcc_s.so.1
11# 0x00007FD9B769C233 in /synopsys/current/customcompiler/linux64/lib/libQt5Core.so.5
12# QMessageLogger::fatal(char const*, ...) const in /synopsys/current/customcompiler/linux64/lib/libQt5Core.so.5
13# QGuiApplicationPrivate::createPlatformIntegration() in /synopsys/current/customcompiler/linux64/lib/libQt5Gui.so.5
14# QGuiApplicationPrivate::createEventDispatcher() in /synopsys/current/customcompiler/linux64/lib/libQt5Gui.so.5
15# QCoreApplicationPrivate::init() in /synopsys/current/customcompiler/linux64/lib/libQt5Core.so.5
16# QGuiApplicationPrivate::init() in /synopsys/current/customcompiler/linux64/lib/libQt5Gui.so.5
17# QApplicationPrivate::init() in /synopsys/current/customcompiler/linux64/lib/libQt5Widgets.so.5
18# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool, bool) in openroad
19# gui::Gui::showGui(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool) in openroad
20# show(char const*, bool, bool) in openroad
21# 0x00000000016DD5A7 in openroad
22# TclNRRunCallbacks in /synopsys/current/customcompiler/linux64/lib/libtcl8.6.so
23# 0x00007FD9BB85C234 in /synopsys/current/customcompiler/linux64/lib/libtcl8.6.so
24# Tcl_EvalEx in /synopsys/current/customcompiler/linux64/lib/libtcl8.6.so
25# Tcl_Eval in /synopsys/current/customcompiler/linux64/lib/libtcl8.6.so
26# 0x0000000000FCF91F in openroad
27# Tcl_MainEx in /synopsys/current/customcompiler/linux64/lib/libtcl8.6.so
28# main in openroad
29# 0x00007FD9B6A295D0 in /lib64/libc.so.6
30# __libc_start_main in /lib64/libc.so.6
31# _start in openroad
Aborted (core dumped)

### Expected Behavior

OpenROAD GUI launches

### Environment

```shell
[root@vlsi01 etc]# ./Env.sh 
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.14.0-570.32.1.el9_6.x86_64
os: Red Hat Enterprise Linux 9.6 (Plow)
cmake version 3.31.6
CMake Error at third-party/CMakeLists.txt:54 (add_subdirectory):
  The source directory

    /usr/openroad/OpenROAD/third-party/abc

[... 149 lines cut ...]
18# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool, bool) in openroad
19# gui::Gui::showGui(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool) in openroad
20# show(char const*, bool, bool) in openroad
21# 0x00000000016DD5A7 in openroad
22# TclNRRunCallbacks in /synopsys/current/customcompiler/linux64/lib/libtcl8.6.so
23# 0x00007FD9BB85C234 in /synopsys/current/customcompiler/linux64/lib/libtcl8.6.so
24# Tcl_EvalEx in /synopsys/current/customcompiler/linux64/lib/libtcl8.6.so
25# Tcl_Eval in /synopsys/current/customcompiler/linux64/lib/libtcl8.6.so
26# 0x0000000000FCF91F in openroad
27# Tcl_MainEx in /synopsys/current/customcompiler/linux64/lib/libtcl8.6.so
28# main in openroad
29# 0x00007FD9B6A295D0 in /lib64/libc.so.6
30# __libc_start_main in /lib64/libc.so.6
31# _start in openroad
Aborted (core dumped)

### Relevant log output

```shell

```

### Screenshots

[build_openroad.log](https://github.com/user-attachments/files/21923732/build_openroad.log)

maliberty, 2025-08-21:
This is most commonly seen if running headless or can't connect to an X server for some other reason.

maliberty, 2025-08-21:
Glad to hear it is working.

closed 2025-08-21 by maliberty