The-OpenROAD-Project/OpenROAD #8101: Resizer: Why seperately track set_dont_touch/set_dont_use and not use the OpenSTA attributes

TheMightyDuckOfDoom, 2025-08-22:
Original review message:

@maliberty should this functionality be moved to odb instead? Seems like purely database related functionality

_Originally posted by @QuantamHD in https://github.com/The-OpenROAD-Project/OpenROAD/pull/8091#discussion_r2292946410_

The Resizer currently has a seperate list for cells with set_dont_touch or set_dont_use attributes.
See https://github.com/The-OpenROAD-Project/OpenROAD/blob/9cc9edfc948c816af5ad03a43194860d4e4664b9/src/rsz/src/Resizer.cc#L2728

Is there a reason why it does not use the OpenSTA attribute/api directly?

gadfort, 2025-08-24:
@QuantamHD I think moving it to ODB would make sense, sometimes you want to be able to save this information between sessions.
I think preserving a separate list from opensta makes sense (that list serves as the starting point right now), but we can overwrite them so different cells and tools become available for different passes.

maliberty, 2025-08-26:
I see the suggestion to move to odb (eg @gadfort comment).  Why do you think dbSta is preferable?

povik, 2025-08-26:
These helpers involve sta types. Can we move those to odb?

povik, 2025-08-26:
This is dont touch and the storage is already in odb, I believe. The helpers are thin wrappers

povik, 2025-08-26:
OK