The-OpenROAD-Project/OpenROAD #8111: Automatic clustering can't be enabled despite message saying so

mole99, 2025-08-25:
### Describe the bug

During CTS I get the following message:

```
[INFO CTS-0204] A clustering solution was found from clustering size of 8 and clustering diameter of 50.
[INFO CTS-0205] Better solution may be possible if either -sink_clustering_size, -sink_clustering_max_diameter, or both options are omitted to enable automatic clustering.
```

Well, I did not pass either `-sink_clustering_size` nor `-sink_clustering_max_diameter`, so what is going on here?

In order for this message to be printed either `clusterSizeSet` or `maxDiameterSet` must be true: https://github.com/The-OpenROAD-Project/OpenROAD/blob/6a23a4b44f9e8b7a8ca3d700bab0e9e87b2dcc57/src/cts/src/HTreeBuilder.cpp#L142

They are set a few lines above by `options_->isMacroMaxDiameterSet()` and `options_->isMacroSinkClusteringSizeSet()` or `options_->isMaxDiameterSet()` and `options_->isSinkClusteringSizeSet()` respectively depending on the TreeType.
(Btw. the datatype of `clusterSizeSet` should be `bool` and not `unsigned`: https://github.com/The-OpenROAD-Project/OpenROAD/blob/6a23a4b44f9e8b7a8ca3d700bab0e9e87b2dcc57/src/cts/src/HTreeBuilder.cpp#L36).

It seems that for a TreeType of `TreeType::MacroTree` the automatic solution is never possible, since `macroSinkClustersSizeSet_` is `true` by default: https://github.com/The-OpenROAD-Project/OpenROAD/blob/6a23a4b44f9e8b7a8ca3d700bab0e9e87b2dcc57/src/cts/src/CtsOptions.h#L319
I think this should be `false`?

Still, why do I get this message for a sink tree?

`isSinkClusteringSizeSet()` is true in my case despite not passing `-sink_clustering_size`, and the only other place that sets it is: https://github.com/The-OpenROAD-Project/OpenROAD/blob/6a23a4b44f9e8b7a8ca3d700bab0e9e87b2dcc57/src/cts/src/TritonCTS.cpp#L214

It seems that my `sinkMaxFanout` equals 8 which is compared with `options_->getSinkClusteringSize()`. Since the latter is larger (`unsigned sinkClustersSize_ = 20;`), `setSinkClusteringSize` is set to 8. But that means that automatic clustering is never possible in my case? If so, the info message should probably not be printed, or the logic should be changed to make this possible.

Does the default value of `sinkClustersSize_ = 20` make sense, when the cluster sizes are as follows?

```
  std::vector<unsigned> clusterSizes_ = {10, 20, 30};
```
(Should the cluster sizes be hardcoded?)

Because as long as `sinkMaxFanout` is below 20, `options_->getSinkClusteringSize()` will be set to `sinkMaxFanout` (without any automatic clustering) as I see it.

Maybe someone who knows what is going on here could take a look to see if I've got something wrong :)

### Expected Behavior

Automatic clustering can be enabled

### Environment

```shell
master branch
```

### To Reproduce

master branch

### Relevant log output

```shell

```

arthurjolo, 2025-08-26:
Indeed CTS is limited in relation to designs with max fanout.

As is the case of the macro tree side. CTS does an automatic selection of cluster diameter based on the macros size, but the cluster size is fixed. As I mentioned the message is misleading. I am making a PR to fix that!

mole99, 2025-08-26:
Yes, but wouldn't it make sense if for a max fanout of 18 the cluster values 10 and 18 are tested?
Again, on what basis were the values in `clusterSizes_` selected?

In my case the message was for clustering sinks and not for macros. It is misleading because my max fanout is too low to enable automatic clustering. My note about `macroSinkClustersSizeSet_` was unrelated to that.

I see now that `-macro_clustering_size` specifies a default value of 4, so please disregard my message about `macroSinkClustersSizeSet_`.

arthurjolo, 2025-08-26:
Yes, it makes sense for a max fanout of 18, to test the cluster values 10 and 18. As I mentioned, CTS is limited in relation to designs with max fanout and I am making a PR to fix that.

arthurjolo, 2025-08-27:
I believe these values were chosen arbitrarily for the designs we had on OpenROAD-flow-scripts. The other alternative is setting the values by hand with `-sink_clustering_size` and `-sink_clustering_max_diameter` as message CTS-0205 tell you not to do 😅 

An improvement on CTS automatic clustering would be to have the cluster sizes and diemeters chosen based on the PDK instead of the arbitrary values.