The-OpenROAD-Project/OpenROAD #8121: drt: Min area is not checked during routing

titan73, 2025-08-26:
### Describe the bug

After routing, I don't see these violations in OR but I see them when doing the DRC checks with Calibre.

Here is 1 example of violations in OR:
[image]
The min area is 0.144 (AREA 0.144 in lef) for layer 2 and the shape is 0.136.

### Expected Behavior

Min area checks should taken into account during detailed routing.

### Environment

```shell
OpenROAD v2.0-23237-g1081cc6d86
```

### To Reproduce

Test case min_area.zip is uploaded to PII sftp server.

### Relevant log output

```shell

```

maliberty, 2025-08-26:
We do check min area so more info is needed to understand the issue.

maliberty, 2025-10-16:
In general the router does not try to repair fixed shapes that were not created by the router.  This case is somewhat at the edge but it would be more simply fixed by just having a pin shape of a legal size.  Is that something you can change?

maliberty, 2025-10-17:
The point is the shape should be legal before being routed.  It is not as it fails min area.  Is there a reason you can't make a legal pin shape?

osamahammad21, 2025-11-19:
I'll close this issue now. Feel free to reopen if needed

closed 2025-11-19 by osamahammad21