The-OpenROAD-Project/OpenROAD #8123: drt: check_drc makes OR crashes

titan73, 2025-08-26:
### Describe the bug

Command drt::check_drc -output drc.txt

Same crash in 2 cases:
  1) after opening an .odb
  2) after the complete flow

### Expected Behavior

Should not crash and generate the report.

### Environment

```shell
OpenROAD v2.0-23770-g72a6f31f7d
```

### To Reproduce

Test cases uploaded to PII sftp server:
   - check_drc.zip for case 1)
   - min_area.zip for case 2)
 

### Relevant log output

```shell
[INFO DRT-0167] List of default vias:
  Layer v1
    default via: M2_M1_VH
  Layer v2
    default via: M3_M2_HV
  Layer v3B
    default via: M4_M3_VH
  Layer padopen
    default via: AP_M4
[INFO DRT-0162] Library cell analysis.
[INFO DRT-0163] Instance analysis.
[INFO DRT-0164] Number of unique instances = 188.
[INFO DRT-0168] Init region query.
[INFO DRT-0024]   Complete poly.
[INFO DRT-0024]   Complete contact.
[INFO DRT-0024]   Complete metal1.
[INFO DRT-0024]   Complete v1.
[INFO DRT-0024]   Complete metal2.
[INFO DRT-0024]   Complete v2.
[INFO DRT-0024]   Complete metal3.
[INFO DRT-0024]   Complete v3B.
[INFO DRT-0024]   Complete metal4B.
[INFO DRT-0024]   Complete padopen.
[INFO DRT-0024]   Complete alucap.
[INFO DRT-0033] poly shape region query size = 0.
[INFO DRT-0033] contact shape region query size = 0.
[INFO DRT-0033] metal1 shape region query size = 824531.
[INFO DRT-0033] v1 shape region query size = 162000.
[INFO DRT-0033] metal2 shape region query size = 98355.
[INFO DRT-0033] v2 shape region query size = 14150.
[INFO DRT-0033] metal3 shape region query size = 7165.
[INFO DRT-0033] v3B shape region query size = 9905.
[INFO DRT-0033] metal4B shape region query size = 4320.
[INFO DRT-0033] padopen shape region query size = 184.
[INFO DRT-0033] alucap shape region query size = 53.
[INFO DRT-0156] guideIn read 100000 guides.
[INFO DRT-0156] guideIn read 200000 guides.
[INFO DRT-0156] guideIn read 300000 guides.
[INFO DRT-0156] guideIn read 400000 guides.

[INFO DRT-0157] Number of guides:     441543

[INFO DRT-0176] GCELLGRID X 0 DO 208 STEP 6150 ;
[INFO DRT-0177] GCELLGRID Y 0 DO 132 STEP 6150 ;
Signal 11 received
Stack trace:
 0# 0x00000000011D1C48 in /prj/gpca_dig/users/costa/openroad/bin/openroad
 1# 0x000014D5BC0835B0 in /lib64/libc.so.6
 2# drt::FlexGCWorker::Impl::checkMetalShape_minArea(drt::gcPin*, bool) in /prj/gpca_dig/users/costa/openroad/bin/openroad
 3# drt::FlexGCWorker::Impl::checkMetalShape_main(drt::gcPin*, bool) in /prj/gpca_dig/users/costa/openroad/bin/openroad
 4# drt::FlexGCWorker::Impl::checkMetalShape(bool) in /prj/gpca_dig/users/costa/openroad/bin/openroad
 5# drt::FlexGCWorker::Impl::main() in /prj/gpca_dig/users/costa/openroad/bin/openroad
 6# 0x0000000002EB29F8 in /prj/gpca_dig/users/costa/openroad/bin/openroad
 7# GOMP_parallel in /sw/freetools/gcc/9.2.0/rh80_64/lib64/libgomp.so.1
 8# drt::TritonRoute::getDRCMarkers(std::__cxx11::list<std::unique_ptr<drt::frMarker, std::default_delete<drt::frMarker> >, std::allocator<std::unique_ptr<drt::frMarker, std::default_delete<drt::frMarker> > > >&, odb::Rect const&) in /prj/gpca_dig/users/costa/openroad/bin/openroad
 9# drt::TritonRoute::checkDRC(char const*, int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int) in /prj/gpca_dig/users/costa/openroad/bin/openroad
10# check_drc_cmd(char const*, int, int, int, int, char const*) in /prj/gpca_dig/users/costa/openroad/bin/openroad
11# 0x0000000001A3593C in /prj/gpca_dig/users/costa/openroad/bin/openroad
12# 0x000014D5C28C9008 in /lib64/libtcl8.6.so
13# TclNRRunCallbacks in /lib64/libtcl8.6.so
14# 0x000014D5C28C7B2F in /lib64/libtcl8.6.so
15# Tcl_EvalEx in /lib64/libtcl8.6.so
16# Tcl_Eval in /lib64/libtcl8.6.so
17# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /prj/gpca_dig/users/costa/openroad/bin/openroad
18# 0x00000000011D147B in /prj/gpca_dig/users/costa/openroad/bin/openroad
19# Tcl_MainEx in /lib64/libtcl8.6.so
20# main in /prj/gpca_dig/users/costa/openroad/bin/openroad
21# __libc_start_main in /lib64/libc.so.6
22# _start in /prj/gpca_dig/users/costa/openroad/bin/openroad
/home/costa/tools/openroad/openroad: line 28: 3426978 Segmentation fault      $root/openroad/bin/openroad $*
```

maliberty, 2025-08-26:
You should use check_drc not the swig wrapper drt::check_drc directly.

maliberty, 2025-08-27:
That is unintentional, fixed in #8130