The-OpenROAD-Project/OpenROAD #8229: Global placement with `-timing_driven` results in critical error

mole99, 2025-09-04:
### Describe the bug

When performing global placement for a full chip design on gf180mcuD, I get the following error:

```
[CRITICAL RSZ-2008] buffering pin gf180mcu_fd_io__in_s_clk/Y: wire step options empty
```

If I omit `-timing_driven`, placement completes successfully.

### Expected Behavior

Global placement should complete

### Environment

```shell
Recent OpenROAD from 2025-09-01 (c47bc3b81431e1ce5f305ab75bad126c3adb2858)
```

### To Reproduce

Here is a standalone reproducible:

[gf180mcu_reproducible.zip](https://github.com/user-attachments/files/22142193/gf180mcu_reproducible.zip)

### Relevant log output

```shell

```

povik, 2025-09-18:
This is an interesting case. The issue is downstream of the IO cell specifying max fanout of 1 in the `.lib`. It will need some adjustment to the buffering algorithm to handle this.

rovinski, 2025-09-19:
If this is a case of the .lib having unrealistic constraints, I think you can ignore it an issue a warning. Even if it's a commercial IP. It seems like the gf180 IO cells are not exactly the best.

mole99, 2025-09-19:
The gf180mcu I/O cells not being the best is an understatement. Turns out despite two separate voltage domains, you cannot have a HV and LV domain due to a lack of level shifters 😆

A warning would be a good enough solution I think.

mole99, 2025-09-26:
Hi, just checking in if a warning could be added with relatively little effort?
I'm currently blocked on this for the [gf180mcu-project-template](https://github.com/wafer-space/gf180mcu-project-template).

povik, 2025-09-30:
> As a workaround, I tried increasing the fanout to 2 in the Liberty files for the I/O cells.

I'm surprised increasing to 2 wasn't enough but there will be some internal reason for it.

> Should OpenROAD not be able to add a buffer directly after the I/O cell output?

That's what it should do but isn't smart enough to do yet. Another workaround would be to insert the buffer into the incoming netlist. It would need a `(* dont_touch *)` attribute to not get removed during optimization steps.

povik, 2025-10-06:
@mole99 I wanted to see why the workaround doesn't work but it seems to work for me. I've used the following script to patch the design in your reproducible.

```
set db [ord::get_db]
set block [ord::get_db_block]
set i 1

foreach pad_inst [$block getInsts] {
	if {[[$pad_inst getMaster] getName] == "gf180mcu_fd_io__in_s" || [[$pad_inst getMaster] getName] == "gf180mcu_fd_io__in_c"} {
		puts "visiting [$pad_inst getName]"
		set buf [$db findMaster gf180mcu_fd_sc_mcu7t5v0__buf_1]
		set buf_in [$buf findMTerm I]
		set buf_out [$buf findMTerm Z]
		set pad_out [[$pad_inst getMaster] findMTerm Y]
		set pad_pin [$pad_inst getITerm $pad_out]
		set n1 [odb::dbNet_create $block "fixup_net$i"]
		set i1 [odb::dbInst_create $block $buf "fixup_buf$i"]
		set i [expr {$i + 1}]
		[$i1 getITerm $buf_in] connect $n1
		[$i1 getITerm $buf_out] connect [$pad_pin getNet]
		$i1 setDoNotTouch true
		$pad_pin connect $n1
	}
}
```

donn, 2025-10-28:
@mole99 In LibreLane it's `librelane.steps create-reproducible` to create the LibreLane reproducible (for us as maintainers) and then `./run_ol.sh eject` to create a shell-only reproducible (to report issues to tools upstream.)

maliberty, 2025-12-17:
I tried your test case and it is also mispackaged:
```
[ERROR ORD-0007] /home/leo/Repositories/ws-submission-2025/example_project/runs/latest/26-odb-customioplacement/user_project_example.odb does not exist.
```
I'm not going to fix again as I did in #8642