The-OpenROAD-Project/OpenROAD #8239: Add yaml-cpp to the DependencyInstaller and update the docker images

osamahammad21, 2025-09-04:
### Description

https://github.com/jbeder/yaml-cpp is used by PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/8214 and is now needed in OpenROAD for parsing yaml format files. We need to add it to the dependency installer and update the docker images used.

vvbandeira, 2025-09-04:
@osamahammad21 
@sombraSoft will coordinate the changes to take care of this; we need to make sure to add to the dependency installer and the dev VMs

osamahammad21, 2025-09-10:
@sombraSoft Any updates on this. It's kind of important and blocking [PR#8214](https://github.com/The-OpenROAD-Project/OpenROAD/pull/8214)

sombraSoft, 2025-09-10:
got this PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/8295 up, sorry for the delay.
I will take a while for it to go through the CI and after it gets merged you just have to make sure you rebase with master and it will work

closed 2025-09-17 by vvbandeira