The-OpenROAD-Project/OpenROAD #8279: Build with RHEL8

Blebowski, 2025-09-09:
### Describe the bug

Hi,
the latest OpenROAD fails to build on RHEL8.

```
[ 49%] Building CXX object src/odb/src/db/CMakeFiles/db.dir/dbBTerm.cpp.o
[ 49%] Building CXX object src/odb/src/db/CMakeFiles/db.dir/dbStream.cpp.o
[ 49%] Building CXX object src/odb/src/db/CMakeFiles/db.dir/dbBTermItr.cpp.o
[ 49%] Building CXX object src/odb/src/db/CMakeFiles/db.dir/dbBPinItr.cpp.o
[ 49%] Building CXX object src/odb/src/db/CMakeFiles/db.dir/dbBlock.cpp.o
/tmp/OpenROAD/src/odb/src/db/dbBlock.cpp: In member function ‘std::__cxx11::string odb::_dbBlock::makeNewName(odb::dbModInst*, const char*, const odb::dbNameUniquifyType&, odb::uint&, const std::function<bool(const char*)>&)’:
/tmp/OpenROAD/src/odb/src/db/dbBlock.cpp:3732:41: error: no matching function for call to ‘fmt::v7::basic_memory_buffer<char>::append(fmt::v7::string_view)’
   buf.append(fmt::string_view(base_name));
                                         ^
In file included from /usr/local/include/spdlog/fmt/fmt.h:22,
                 from /usr/local/include/spdlog/common.h:36,
                 from /usr/local/include/spdlog/details/os.h:6,
                 from /tmp/OpenROAD/src/utl/include/utl/Logger.h:28,
                 from /tmp/OpenROAD/src/odb/include/odb/odb.h:15,
                 from /tmp/OpenROAD/src/odb/include/odb/ZException.h:6,
                 from /tmp/OpenROAD/src/odb/src/db/dbAttrTable.h:6,
                 from /tmp/OpenROAD/src/odb/src/db/dbCore.h:28,
                 from /tmp/OpenROAD/src/odb/src/db/dbBlock.h:13,
                 from /tmp/OpenROAD/src/odb/src/db/dbBlock.cpp:4:
/usr/local/include/spdlog/fmt/bundled/core.h:700:30: note: candidate: ‘template<class U> void fmt::v7::detail::buffer<T>::append(const U*, const U*) [with U = U; T = char]’
   template <typename U> void append(const U* begin, const U* end);
                              ^~~~~~
/usr/local/include/spdlog/fmt/bundled/core.h:700:30: note:   template argument deduction/substitution failed:
/tmp/OpenROAD/src/odb/src/db/dbBlock.cpp:3732:41: note:   mismatched types ‘const U*’ and ‘fmt::v7::basic_string_view<char>’
   buf.append(fmt::string_view(base_name));
                                         ^
make[2]: *** [src/odb/src/db/CMakeFiles/db.dir/build.make:132: src/odb/src/db/CMakeFiles/db.dir/dbBlock.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2567: src/odb/src/db/CMakeFiles/db.dir/all] Error 2
```

### Expected Behavior

To compile without problems.

I know RHEL8 is a legacy OS now, but we are still not migrating since SNPS guarantees RHEL9 support from releases in Q2 2026, so we will have some transition period in the nearest year. It would be very nice to have OR support this.

### Environment

```shell
Git commit: 713a21292340b8b187748ad728cdf1384543d466
kernel: Linux 4.18.0-553.58.1.el8_10.x86_64
os: Red Hat Enterprise Linux 8.10 (Ootpa)
cmake version 3.26.5
-- The CXX compiler identification is GNU 8.5.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-24267-g713a212923
-- System name: Linux
-- Compiler: GNU 8.5.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Older version of GCC detected. Linking against stdc++fs
-- The C compiler identification is GNU 8.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
[... 131 lines cut ...]
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found OpenMP: TRUE (found version "4.5")  
-- QT5 not found. GUI is disabled
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done (6.7s)
-- Generating done (0.5s)
-- Build files have been written to: /tmp/tmp.Jj5dKlw0c7
```

### To Reproduce

-

### Relevant log output

```shell

```

vvbandeira, 2025-09-09:
We are able to build on RHEL8, but it does require a more modern GCC than the one you have.
Our CI uses GCC 13, so this version is known to work, but 9+ should work as well.
You can install with `sudo dnf install gcc-toolset-13`

Blebowski, 2025-09-24:
Hi @vvbandeira , I am using the DependencyInstaller.sh. I can see it downloads the Bison 3.8.4, but for some reason it is not taken into account when configuring the "swig" build, nor the OR build:
```
echo "/* SWIG warning codes - generated from swigwarn.h - do not edit */" > Lib/swigwarn.swg
cat Source/Include/swigwarn.h | grep "^#define WARN\|/\*.*\*/\|^[ \t]*$" | sed 's/^#define \(WARN.*[0-9][0-9]*\)\(.*\)$/%define SWIG\1 %enddef\2/' >> Lib/swigwarn.swg
make[1]: Entering directory '/tmp/DependencyInstaller-Gb5rDn/swig-4.3.0/CCache'
gcc -g -O2 -Wall -W -I.   -c -o ccache.o ccache.c
gcc -g -O2 -Wall -W -I.   -c -o mdfour.o mdfour.c
gcc -g -O2 -Wall -W -I.   -c -o hash.o hash.c
make[1]: Entering directory '/tmp/DependencyInstaller-Gb5rDn/swig-4.3.0/Source'
bison -d -Wall -Werror  --output=CParse/parser.c ./CParse/parser.y
gcc -g -O2 -Wall -W -I.   -c -o execute.o execute.c
gcc -g -O2 -Wall -W -I.   -c -o util.o util.c
./CParse/parser.y:16.10-14: error: require bison 3.5, but have 3.0.4
 %require "3.5"
          ^^^^^
make[1]: *** [Makefile:1363: CParse/parser.c] Error 63
make[1]: Leaving directory '/tmp/DependencyInstaller-Gb5rDn/swig-4.3.0/Source'
make: *** [Makefile:36: source] Error 2
make: *** Waiting for unfinished jobs....
```
I will try to uninstall the default system version of `bison` and try then.