The-OpenROAD-Project/OpenROAD #8365: bazel build currently fails: `src/pdn/src/PdnGen-py.i:23: Error: Unable to find 'odb.i'`

hzeller, 2025-09-18:
### Describe the bug

(build issue. Not sure how well that fits into regular 'bug report')

Trying to build the project with bazel currently fails:

```
$ bazel build -c opt src/...
ERROR: /home/hzeller/src/my/OpenROAD/src/ppl/BUILD:100:15: Action src/ppl/swig-py.cc failed: (Exit 1): swig failed: error executing Action command (from target //src/ppl:swig-py) bazel-out/k8-opt-exec-ST-d57f47055a04/bin/external/org_swig/swig '-DBAZEL=1' -python -c++ -flatstaticmethod -module ppl_py -Isrc/ppl/include -Isrc/ppl/src -o bazel-out/k8-opt/bin/src/ppl/swig-py.cc ... (remaining 1 argument skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
src/ppl/src/IOPlacer-py.i:42: Error: Unable to find 'odb.i'
ERROR: /home/hzeller/src/my/OpenROAD/src/pdn/BUILD:95:15: Action src/pdn/swig-py.cc failed: (Exit 1): swig failed: error executing Action command (from target //src/pdn:swig-py) bazel-out/k8-opt-exec-ST-d57f47055a04/bin/external/org_swig/swig '-DBAZEL=1' -python -c++ -flatstaticmethod -module pdn_py -Isrc/pdn/include -o bazel-out/k8-opt/bin/src/pdn/swig-py.cc ... (remaining 1 argument skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
src/pdn/src/PdnGen-py.i:23: Error: Unable to find 'odb.i'
```

Is this known and due to WIP on the Python integration @QuantamHD @maliberty ?

### Expected Behavior

Build completes :)

### Environment

```shell
bazel --version
bazel 8.3.1
```

### To Reproduce

```
bazel build -c opt src/...
```

### Relevant log output

```shell

```

QuantamHD, 2025-09-18:
I think it's because this rule does not build. It probably needs to take a dep on odb.i. The root openroad rule doesn't actually run swig on this file individually so the error won't show up unless you build this rule specifically. 

```

python_wrap_cc(
    name = "swig-py",
    srcs = [
        "include/ppl/Parameters.h",
        "include/ppl/IOPlacer.h",
        "src/IOPlacer-py.i",
        "//:error_swig-py",
    ],
    module = "ppl_py",
    root_swig_src = "src/IOPlacer-py.i",
    swig_includes = [
        "include",
        "src",
    ]
)
```

hzeller, 2025-09-21:
`bazel test -c opt src/...` works, but `bazel build -c opt src/...` does not. Some not entirely clean dependency declaration might be the issue.