The-OpenROAD-Project/OpenROAD #8373: [ERROR PDN-0233] Failed to generate full power grid.

oharboe, 2025-09-19:
### Describe the bug

[pdn_top_asap7_base_2025-09-19_04-03.tar.gz](https://github.com/user-attachments/files/22417801/pdn_top_asap7_base_2025-09-19_04-03.tar.gz)

```
INFO PDN-0001] Inserting grid: top
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - u_sdq_17x64
[WARNING PDN-0232] CORE_macro_grid_1 - u_sdq_17x64 does not contain any shapes or vias.
[ERROR PDN-0233] Failed to generate full power grid.
Error: pdn.tcl, 6 PDN-0233
```

### Expected Behavior

No error

### Environment

```shell
OpenROAD 24Q3-8871-g96512a37e9
```

### To Reproduce

See above

### Relevant log output

```shell

```

oharboe, 2025-09-19:
@maliberty @vvbandeira Funky version number still in Docker...

maliberty, 2025-09-19:
You have
```
define_pdn_grid -name {CORE_macro_grid_1} -voltage_domains {CORE} -macro \
  -orient {R0 R180 MX MY} -halo {2.0 2.0 2.0 2.0} -cells {.*}
add_pdn_connect -grid {CORE_macro_grid_1} -layers {M4 M5}
```
but your block has shapes/blockages up to M6 so M4/M5 isn't going to work.

oharboe, 2025-09-19:
> You have
> 
> ```
> define_pdn_grid -name {CORE_macro_grid_1} -voltage_domains {CORE} -macro \
>   -orient {R0 R180 MX MY} -halo {2.0 2.0 2.0 2.0} -cells {.*}
> add_pdn_connect -grid {CORE_macro_grid_1} -layers {M4 M5}
> ```
> 
> but your block has shapes/blockages up to M6 so M4/M5 isn't going to work.

I'm using ASAP7 default PDN_TCL, so https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/bb27eaa879385f5ae08e008e43675f6ba1065a76/flow/platforms/asap7/config.mk#L42

oharboe, 2025-09-19:
Does https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/bb27eaa879385f5ae08e008e43675f6ba1065a76/flow/platforms/asap7/config.mk#L42 need an update?

This works:

```
$ git diff
diff --git a/sram/BUILD b/sram/BUILD
index 10d1a0a..9743728 100644
--- a/sram/BUILD
+++ b/sram/BUILD
@@ -22,6 +22,7 @@ FAST_SETTINGS = {
         "CORE_MARGIN": "2",
         "MACRO_PLACE_HALO": "30 30",
         "PLACE_DENSITY": "0.23",
+        "PDN_TCL": "$(PLATFORM_DIR)/openRoad/pdn/BLOCK_grid_strategy.tcl"
     },
     mock_area = 0.95,
     previous_stage = {
```

maliberty, 2025-09-20:
gcd probably can route everything on the lower layers.

oharboe, 2025-09-20:
> gcd probably can route everything on the lower layers.

How does pdn.tcl know that everything can be routed on the lower layers?

This is all a bit mysterious to me. It sounds to me that PDN is working as it should, but perhaps not after the intention of the default ASAP7 [PDN_TCL](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/bb27eaa879385f5ae08e008e43675f6ba1065a76/flow/platforms/asap7/config.mk#L42)

What should I expect from the default PDN_TCL from ASAP7 if routing needs the uppper layers?

I'm missing some important information and mental model for this to all make sense... Also, if the default PDN_TCL for ASAP7 needs to be "fixed" (whatever that means), I don't know how to do that.

oharboe, 2025-09-20:
One more testcase, created from bazelisk build //sram:top_floorplan https://github.com/The-OpenROAD-Project/bazel-orfs/pull/380

The submacro uses https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/asap7/openRoad/pdn/BLOCK_grid_strategy.tcl  and the top level https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/asap7/openRoad/pdn/BLOCKS_grid_strategy.tcl

Untar  [pdn-fail.tar.gz](https://github.com/user-attachments/files/22448168/pdn-fail.tar.gz) and run:

    make --file=~/OpenROAD-flow-scripts/flow/Makefile DESIGN_CONFIG=config.mk do-floorplan

```
[deleted]
[INFO PDN-0001] Inserting grid: top
[INFO PDN-0001] Inserting grid: ElementGrid - u_sdq_17x64
[WARNING PDN-0232] ElementGrid - u_sdq_17x64 does not contain any shapes or vias.
[ERROR PDN-0233] Failed to generate full power grid.
Error: pdn.tcl, 6 PDN-0233
```

oharboe, 2025-09-21:
Near as I can tell, this is fixed by using BLOCK and  BLOCKS PDN filed from ASAP7.

The takeaway here is that a more actionable easier to understand error message would be good here.

Possible?

gadfort, 2025-09-21:
@oharboe I'm open to concrete suggestions for how the message could be modified