The-OpenROAD-Project/OpenROAD #8380: Add "Routing delay" column to compliment "Logic delay" column

oharboe, 2025-09-20:
### Description

I'd like to have a "Routing delay" column to compliment the "Logic delay" column, this is useful when primarily looking at timing reports to understand floorplan/routing delay issues.

[image]

### Suggested Solution

Add "Routing delay" column in Timing Report.

"Routing delay" & "Logic delay" in a dropdown for the "Endpoint slack histogram" would be nice too. At which point the "Logic depth", "Fanout" endpoint histograms are obvious followup asks. Adding a dropdown is probably trivial, but is this information easily acccessible? I seem to recall that Endpoint slack histogram shows the worst slack for each endpoint and that this is harder than it sounds. Can this be done for the worst slack path and used in the histogram? So it would still be an "Endpoint worst slack histogram", but showing the histogram for other properties of the worst slack path, not just the worst slack.

maliberty, 2025-09-22:
Logic delay excludes buffers and inverter pairs so would "Routing delay" include them?  If so it could be a confusing name as it sounds like it is just wiring delay.

oharboe, 2025-09-22:
> Logic delay excludes buffers and inverter pairs so would "Routing delay" include them? If so it could be a confusing name as it sounds like it is just wiring delay.

How about "Non-Logic delay"?

I really just want the compliment of the Logic Delay rather than having to do mental math when reading the table and to be able to sort on "Non-Logic delay".