The-OpenROAD-Project/OpenROAD #8416: [bazel] _openroadpy.so: undefined symbol: _ZSt21__glibcxx_assert_failPKciS0_S0_

oharboe, 2025-09-24:
### Describe the bug

Run:

    bazelisk test //src/odb/test:test_module --test_output=errors

```
INFO: Invocation ID: 26532326-86a2-4428-ac95-fc7912f46ae1
INFO: Analyzed target //src/odb/test:test_module (0 packages loaded, 0 targets configured).
FAIL: //src/odb/test:test_module (Exit 1) (see /home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/execroot/_main/bazel-out/k8-fastbuild/testlogs/src/odb/test/test_module/test.log)
INFO: From Testing //src/odb/test:test_module:
==================== Test output for //src/odb/test:test_module:
Traceback (most recent call last):
  File "/home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/sandbox/processwrapper-sandbox/8036/execroot/_main/bazel-out/k8-fastbuild/bin/src/odb/test/test_module.runfiles/_main/src/odb/test/test_module.py", line 2, in <module>
    import helper
  File "/home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/sandbox/processwrapper-sandbox/8036/execroot/_main/bazel-out/k8-fastbuild/bin/src/odb/test/test_module.runfiles/_main/src/odb/test/helper.py", line 2, in <module>
    import openroad
  File "/home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/sandbox/processwrapper-sandbox/8036/execroot/_main/bazel-out/k8-fastbuild/bin/src/odb/test/test_module.runfiles/_main/openroad/__init__.py", line 4, in <module>
    from openroadpy import *
  File "/home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/sandbox/processwrapper-sandbox/8036/execroot/_main/bazel-out/k8-fastbuild/bin/src/odb/test/test_module.runfiles/_main/openroadpy.py", line 12, in <module>
    import _openroadpy
ImportError: /home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/sandbox/processwrapper-sandbox/8036/execroot/_main/bazel-out/k8-fastbuild/bin/src/odb/test/test_module.runfiles/_main/_openroadpy.so: undefined symbol: _ZSt21__glibcxx_assert_failPKciS0_S0_
================================================================================
INFO: Found 1 test target...
Target //src/odb/test:test_module up-to-date:
  bazel-bin/src/odb/test/test_module
INFO: Elapsed time: 2.496s, Critical Path: 2.14s
INFO: 2 processes: 2 processwrapper-sandbox.
INFO: Build completed, 1 test FAILED, 2 total actions
//src/odb/test:test_module                                               FAILED in 0.3s
  /home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/execroot/_main/bazel-out/k8-fastbuild/testlogs/src/odb/test/test_module/test.log

Executed 1 out of 1 test: 1 fails locally.
```

### Expected Behavior

No error

### Environment

```shell
v2.0-24770-g646a606ef2

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 25.04
Release:	25.04
Codename:	plucky
```

### To Reproduce

See above, on Ubuntu

### Relevant log output

```shell

```

QuantamHD, 2025-09-26:
Can you run ldd on the _openroad.so object and paste what it returns.

oharboe, 2025-09-26:
```
$ ldd /home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/sandbox/processwrapper-sandbox/725/execroot/_main/bazel-out/k8-fastbuild/bin/src/odb/test/test_module.runfiles/_main/_openroadpy.so
	linux-vdso.so.1 (0x000073cab31f6000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x000073caa4d09000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x000073caa4a00000)
	/lib64/ld-linux-x86-64.so.2 (0x000073cab31f8000)
```

oharboe, 2025-09-26:
```
$ bazelisk test //src/odb/test:test_module --test_output=errors --sandbox_debug
[deleted]
ImportError: /home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/sandbox/processwrapper-sandbox/8/execroot/_main/bazel-out/k8-fastbuild/bin/src/odb/test/test_module.runfiles/_main/_openroadpy.so: undefined symbol: _ZSt21__glibcxx_assert_failPKciS0_S0_
[deleted]
$ git diff
diff --git a/BUILD.bazel b/BUILD.bazel
index 7f01c821df..fed3f34b1c 100644
--- a/BUILD.bazel
+++ b/BUILD.bazel
@@ -281,6 +281,7 @@ cc_binary(
         "@boost.stacktrace",
         "@openroad_rules_python//python/cc:current_py_cc_headers",
     ],
+    linkstatic = True,
 )
 
 # This packages the SWIG-generated Python wrapper (odb.py) and the
$ ldd /home/oyvind/.cache/bazel/_bazel_oyvind/896cc02f64446168f604c13ad7b60f8b/sandbox/processwrapper-sandbox/8/execroot/_main/bazel-out/k8-fastbuild/bin/src/odb/test/test_module.runfiles/_main/_openroadpy.so
	linux-vdso.so.1 (0x00007dfccd9c6000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007dfcbf509000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007dfcbf200000)
	/lib64/ld-linux-x86-64.so.2 (0x00007dfccd9c8000)
$ git log -1
commit 774459f710ba7b9ae0506865cd0af242090c1a9f (HEAD, origin/master, origin/HEAD)
Merge: 792fcced6d 3739b64619
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Thu Sep 25 22:51:46 2025 +0000

    Merge pull request #8439 from osamahammad21/3dblox-paths-defines
    
    ODB-3Dblox: Defines and paths
```

maliberty, 2025-09-30:
No error with 9103e99444dc8aedf2ce36b5e131ba4230590a56 and Ubuntu 24.  We don't support the non-LTS releases so I don't know if 25 is the issue.

oharboe, 2025-09-30:
Ubuntu 25.10 is scheduled for release Oct 9. and we don't use the python support, so for me it is no problem re-trying in two weeks or so.

maliberty, 2025-09-30:
Note that only the even numbered releases are LTS

mattyoung101, 2025-10-07:
In my case, I've been using the following command to build successfully:

```
bazelisk build -c opt //:openroad --//:platform=gui
```

I can also confirm `bazelisk test -c opt` works. I think the `-c opt` will define `-DNDEBUG` or such and mean that the asserts don't get compiled in. So this is specifically a problem in debug builds, which I guess are the default.

QuantamHD, 2025-10-07:
I've experimented a bit and it seems many python things pass -Bsymbolic and or -Bsymbolic-functions to the linker options that might be a thing to try