The-OpenROAD-Project/OpenROAD #8466: grt: Global routing very slow (Regression due to 5ca183346a58f00bb242116caaff7e6ecce53934)

titan73, 2025-09-29:
### Describe the bug

The following commit makes the global routing very slow (3mn before and 20h after):

```
commit 5ca183346a58f00bb242116caaff7e6ecce53934
Author: Jonas Gava <jfgava@precisioninno.com>
Date:   Wed Sep 10 22:42:29 2025 +0000

    cts: enable half-NDR for clock nets by default. Fix clock tree level list check.
    
    Signed-off-by: Jonas Gava <jfgava@precisioninno.com>
```

There are a lot of messages related to NDR:
```
INFO GRT-0101] Running extra iterations to remove overflow.
[WARNING GRT-0273] Disabled NDR for the clknet_0_STUSB_ISO_i\/clock_shop_clk_32k net.
[WARNING GRT-0273] Disabled NDR for the clknet_0_STUSB_ISO_i\/clock_shop_stusb_i_o_clk_i2c net.
[WARNING GRT-0273] Disabled NDR for the clknet_0_STUSB_ISO_i\/clock_shop_stusb_i_o_clk_i2c_regs net.
[WARNING GRT-0273] Disabled NDR for the clknet_0_STUSB_ISO_i\/registers_i\/net22290 net.
[WARNING GRT-0273] Disabled NDR for the clknet_0_STUSB_ISO_i\/registers_i\/clk_int net.
[WARNING GRT-0273] Disabled NDR for the STUSB_ISO_i/registers_i/net22160 net.
[WARNING GRT-0273] Disabled NDR for the STUSB_ISO_i/clock_shop_clk_32k net.
[WARNING GRT-0273] Disabled NDR for the STUSB_ISO_i/registers_i/net22170 net.
[WARNING GRT-0273] Disabled NDR for the clknet_0_STUSB_ISO_i\/clock_shop_clk_32k_regs net.
[WARNING GRT-0273] Disabled NDR for the clknet_0_STUSB_ISO_i\/n739 net.
```

There are no NDR created in the design.

### Expected Behavior

Should not be slow without any NDR related messages.

### Environment

```shell
RHEL 8
```

### To Reproduce

Test case `grt_slow_ndr.zip` uploaded to PII sftp server.

### Relevant log output

```shell

```

maliberty, 2025-09-29:
Can you provide a test case?  If not you could just disabled the NDR generation for your design.

jfgava, 2025-10-01:
> With `-apply_ndr none`, things are back to normal. So I understand there are default NDRs? What are they?

Yes, now we have 2s1w NDR applied to some clock nets by default. For these nets, the spacing between wires is doubled. 
In general, it provides a significant improvement in timing. But in some cases, it can generate congestion issues

titan73, 2025-10-02:
Ok thanks. And what does mean the message?
`[WARNING GRT-0273] Disabled NDR for the clknet_0_STUSB_ISO_i\/clock_shop_clk_32k net.`
Why were the NDR disabled for this net (and all the others)?

titan73, 2025-10-02:
Ok thanks for the explaination.

Maybe I would suggest to improve the message like "Disabled NDR to reducing congestion for the ...".