The-OpenROAD-Project/OpenROAD #8469: Reduce PDN spam

oharboe, 2025-09-29:
### Description

untar and run  

```
[WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_1): A2O1A1Ixp33_ASAP7_75t_R
[WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_1): A2O1A1O1Ixp25_ASAP7_75t_R
[WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_1): AND2x2_ASAP7_75t_R
[WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_1): AND2x4_ASAP7_75t_R
[WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_1): AND2x6_ASAP7_75t_R
[WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_1): AND3x1_ASAP7_75t_R
[WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_1): AND3x2_ASAP7_75t_R
[WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_1): AND3x4_ASAP7_75t_R
[WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_1): AND4x1_ASAP7_75t_R
[WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_1): AND4x2_ASAP7_75t_R
[WARNING PDN-1041] Ignoring non-block cell for grid (CORE_macro_grid_1): AND5x1_ASAP7_75t_R
```

### Suggested Solution

Actionable warnings only

gadfort, 2025-09-30:
consider replacing
```
define_pdn_grid -name {CORE_macro_grid_1} -voltage_domains {CORE} -macro \
  -orient {R0 R180 MX MY} -halo {2.0 2.0 2.0 2.0} -cells {.*}
define_pdn_grid -name {CORE_macro_grid_2} -voltage_domains {CORE} -macro \
  -orient {R90 R270 MXR90 MYR90} -halo {2.0 2.0 2.0 2.0} -cells {.*}
```
with
```
define_pdn_grid -name {CORE_macro_grid_1} -voltage_domains {CORE} -macro \
  -orient {R0 R180 MX MY} -halo {2.0 2.0 2.0 2.0} -default
define_pdn_grid -name {CORE_macro_grid_2} -voltage_domains {CORE} -macro \
  -orient {R90 R270 MXR90 MYR90} -halo {2.0 2.0 2.0 2.0} -default
```

Note that CORE_macro_grid_2 is extremely unlikely to be useful or result in valid transistor orientations

maliberty, 2025-10-01:
I don't think there is any harm in that rule.  ng45 is non-manufacturable so if someone wants to play with rotated macros I don't see a reason to not support it.

closed 2025-10-01 by maliberty