The-OpenROAD-Project/OpenROAD #8473: [ERROR PDN-0006] Pins on Metal4 are blocked by obstructions on Metal5 for system_expander.i2cCtrl

eder-matheus, 2025-09-30:
### Describe the bug

PDN step fails with error `[ERROR PDN-0006] Pins on Metal4 are blocked by obstructions on Metal5 for system_expander.i2cCtrl` when using the latest version of Yosys.

This is currently blocking PR https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/3530.

### Expected Behavior

PDN finishes with no errors.

### Environment

```shell
OpenROAD version: v2.0-24945-g9103e99444
Yosys version: Yosys 0.57+218 (git sha1 30cb72a16, g++ 13.3.0-6ubuntu2~24.04 -fPIC -O3)
```

### To Reproduce

```
tar -xf pdn_issue.tar.gz
cd pdn_issue
./run-me-i2c-gpio-expander-ihp-sg13g2-base.sh
```

[pdn_issue.tar.gz](https://github.com/user-attachments/files/22619980/pdn_issue.tar.gz)

### Relevant log output

```shell

```

eder-matheus, 2025-09-30:
Extra information: the instance reported in the error message has a stack of obstructions from Metal5 to Metal1, as shown in the figure below.
[image]

The obstructions are blocking the whole instance. Is this expected?

eder-matheus, 2025-09-30:
The master branch result has obstructions up to Metal4:
[image]

It seems that the PDN is not the cause of the issue, but it's just showing a problem from previous steps.

maliberty, 2025-09-30:
If there are any shapes on metal5 then the abstract will block the full layer.  The problem is that when running I2cDeviceCtrl the MAX_ROUTING_LAYER needs to be set lower in /designs/ihp-sg13g2/i2c-gpio-expander/I2cDeviceCtrl/config.mk